project-chip/connectedhomeip #28604: [CERT-TEST-FAILURE] - TC-BR-4 - Test failed

Survensa, 2023-08-09:
### Feature Area

Other

### Test Case

TC-BR-4

### Reproduction steps

1. Bring up the DUT with bridge-app.
2. Pair the DUT using chip-tool through "onnetwork" transport.
3. Fan is set to medium mode `./chip-tool powersource read bat-charge-level 1 9`

**Expected Outcome:**

> [1663925822.845859][1588:1588] CHIP:DMG:                                         Endpoint = 0x9,
> [1663925822.845888][1588:1588] CHIP:DMG:                                         Cluster = 0x2f,
> [1663925822.845919][1588:1588] CHIP:DMG:                                         Attribute = 0x0000_000E,
> [1663925822.845946][1588:1588] CHIP:DMG:                                 }
> [1663925822.845978][1588:1588] CHIP:DMG:                                         
> [1663925822.846007][1588:1588] CHIP:DMG:                                 Data = 58, 
> [1663925822.846035][1588:1588] CHIP:DMG:                         },

**Actual Outcome:**

> 691582024.386805][3477:3479] CHIP:DMG: 					Endpoint = 0x9,
> [1691582024.386853][3477:3479] CHIP:DMG: 					Cluster = 0x2f,
> [1691582024.386901][3477:3479] CHIP:DMG: 					Attribute = 0x0000_000E,
> [1691582024.386946][3477:3479] CHIP:DMG: 				}
> [1691582024.386998][3477:3479] CHIP:DMG: 					
> [1691582024.387039][3477:3479] CHIP:DMG: 				StatusIB =
> [1691582024.387084][3477:3479] CHIP:DMG: 				{
> [1691582024.387129][3477:3479] CHIP:DMG: 					status = 0xc3 (UNSUPPORTED_CLUSTER),
> [1691582024.387174][3477:3479] CHIP:DMG: 				},

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

367a0c672c18e0137e0162c15b78adbcb039a33a

### Platform

raspi

### Anything else?

Seems it broke after the PR: https://github.com/project-chip/connectedhomeip/pull/28134

andy31415, 2023-08-29:
@Survensa it seems that a bridge is not mandated to have a power source. Test plans can validate power source clusters if it exists, however it probably needs to determine first if power source is supported or not.

Could we get some details about what the test plan is expecting?

Survensa, 2023-08-30:
@andy31415

Test Plan expectations:

- Verify whether the Device Under Test (DUT) successfully reads the BatChargeLevel attribute from the Power Source cluster, either from the appropriate endpoint or by receiving updates due to a pre-established subscription.
- Confirm that the DUT accurately maintains the battery state of the supported devices listed above.

Note: Prior to the merge of PR: https://github.com/project-chip/connectedhomeip/pull/28134 into the SDK, this step obtains an expected value from the SDK.

andy31415, 2023-08-30:
What guarantees that "endpoint 9" must have a power source cluster?

andy31415, 2023-08-30:
Manual checking bridge app seems to indicate to me that cluster 47 is available on endpoint 6 ... however I do not actually know what each of these endpoints are supposed to represent. What are the test assumptions?

andy31415, 2023-08-30:
```
> ./out/linux-x64-chip-tool/chip-tool powersource read  bat-charge-level 1 6 --trace-to json:log
....
[1693417708.849289][1253442:1253444] CHIP:ATM:          {                                                                                      [105/94601]
[1693417708.849292][1253442:1253444] CHIP:ATM:                  "report_data" :
[1693417708.849296][1253442:1253444] CHIP:ATM:                  {
[1693417708.849300][1253442:1253444] CHIP:ATM:                          "attribute_reports" :
[1693417708.849303][1253442:1253444] CHIP:ATM:                          {
[1693417708.849307][1253442:1253444] CHIP:ATM:                                  "Anonymous<0>" :
[1693417708.849311][1253442:1253444] CHIP:ATM:                                  {
[1693417708.849314][1253442:1253444] CHIP:ATM:                                          "attribute_data" :
[1693417708.849318][1253442:1253444] CHIP:ATM:                                          {
[1693417708.849322][1253442:1253444] CHIP:ATM:                                                  "PowerSource::batChargeLevel" : "58",
[1693417708.849327][1253442:1253444] CHIP:ATM:                                                  "data_version" : "1622213645",
[1693417708.849331][1253442:1253444] CHIP:ATM:                                                  "path" :
[1693417708.849335][1253442:1253444] CHIP:ATM:                                                  {
[1693417708.849339][1253442:1253444] CHIP:ATM:                                                          "attribute_id" : "14 == 'batChargeLevel'",
[1693417708.849343][1253442:1253444] CHIP:ATM:                                                          "cluster_id" : "47 == 'PowerSource'",
[1693417708.849348][1253442:1253444] CHIP:ATM:                                                          "endpoint_id" : "6"
[1693417708.849352][1253442:1253444] CHIP:ATM:                                                  }
[1693417708.849356][1253442:1253444] CHIP:ATM:                                          }
[1693417708.849359][1253442:1253444] CHIP:ATM:                                  }
[1693417708.849363][1253442:1253444] CHIP:ATM:                          },
[1693417708.849370][1253442:1253444] CHIP:ATM:                          "interaction_model_revison" : "10",
[1693417708.849373][1253442:1253444] CHIP:ATM:                          "suppress_response" : "true"
[1693417708.849377][1253442:1253444] CHIP:ATM:                  }
[1693417708.849383][1253442:1253444] CHIP:ATM:          },
```

so this does work for that endpoint, however it seems unclear to me how we decide what endpoint to test for.

andy31415, 2023-08-30:
According to the source comments in https://github.com/project-chip/connectedhomeip/blob/master/examples/bridge-app/linux/main.cpp#L967 the existing code expects:

- endpoint 3 : a light
- endpoint 4,5: temperature sensors
- endpoint 6: composed device with temperature sensor and power source (It seems want to test this one!) 
   - call wise it looks to me like we have endpoint 6,7,8 being added here
-  endpoints 9,10,11,12 are lights that are stated to be for "action clusters tests"

andy31415, 2023-08-30:
#28134 seems to have removed a previous "endpoint 9" and kept the power source on endpoint 6.

@cecille may know the spec correctness here, but from a test perspective it looks like this test needs to validate endpoint 6 if all we need is to read battery level

Survensa, 2023-08-31:
Thank you @andy31415

closed 2023-08-31 by Survensa