project-chip/connectedhomeip #28605: [BUG] [TE2- v1.2] Descriptor cluster's TagList feature is not implemented in the SDK but present in the specs for v1.2

PSONALl, 2023-08-09:
### Reproduction steps

Descriptor cluster's TagList feature is not implemented in the SDK but present in the specs for v1.2

### Bug prevalence

EveryTime

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372 

### Platform

core

### Platform Version(s)

v1.2

bzbarsky-apple, 2023-08-09:
@PSONALl Is this a duplicate of https://github.com/project-chip/connectedhomeip/issues/28389 or https://github.com/project-chip/connectedhomeip/issues/28098 ?

PSONALl, 2023-08-10:
> @PSONALl Is this a duplicate of #28389 or #28098 ?

Yes, I think the issue's cause might be the same. But the thing to be noted is that along with the TagList Attribute, TagList Feature is also not defined in cluster-enums.h

bzbarsky-apple, 2023-08-10:
Ah, good point.  If this issue is about the missing XML bit, fix coming up.

closed 2023-08-11 by mergify[bot]