project-chip/connectedhomeip #28611: [BUG]  facing error on building all-clusters-app for esp32

aravindone, 2023-08-09:
### Reproduction steps

steps to recreate.

1) activate the esp-idf v5.1 environment.
2) activate the  connectedhomeip environment.
3)move to examples/all-clusters-app/esp32.
4)idfy.py set-target esp32 (working fine).
5)idf.py build (problem arises).

attached log : 

[error_all-cluster-app.TXT](https://github.com/project-chip/connectedhomeip/files/12305271/error_all-cluster-app.TXT)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

367a0c672c18e0137e0162c15b78adbcb039a33a

### Platform

esp32

### Anything else?

@shubhamdp  @dhrishi @bzar

bzbarsky-apple, 2023-08-09:
@aravindone On https://github.com/project-chip/connectedhomeip/commit/367a0c672c18e0137e0162c15b78adbcb039a33a there are no mentions of `Esp32ThreadInit.h` anywhere in the tree.  What SHA are you _actually_ compiling?

bzbarsky-apple, 2023-08-09:
I did just try on d7414cc29442b3f29f9b978a67421a138f9c8861 (but without running `idfy.py set-target esp32`) and things work fine for me.

aravindone, 2023-08-10:
@bzbarsky-apple  then shall i update my repo to https://github.com/project-chip/connectedhomeip/commit/d7414cc29442b3f29f9b978a67421a138f9c8861 and check .

aravindone, 2023-08-10:
hi @bzbarsky-apple , i uploaded to the latest commit now works fine a3c4b483651b1f5a44039da96857deed9122860f.

thanks..

closed 2023-08-10 by aravindone