project-chip/connectedhomeip #28621: [BUG] update the data of attributes using custom.cpp script

aravindone, 2023-08-10:
### Reproduction steps

Dear devs  i need to know how to push the data got by reading from gpio or virtually the file to their respective attributes on an ESP 32

1) How to write custom .cpp file in all clusters app project space. 
2) from that custom.cpp read data from the esp 32 virtually.
3) want to access the electrical measurement class to set the value.
4) update the data in total apparent power attribute from electrical measurement cluster.

problem i faced .
i cant able to alter the main.cpp file of esp32 example as many error are popping up while compiling.

need guidance for above implementation.
like what are the alterations i need to make  in files after developing custom script to work with main.cpp program.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

367a0c672c18e0137e0162c15b78adbcb039a33a

### Platform

esp32

### Anything else?

@dhrishi @shubhamdp @bzbarsky-apple  please comment

bzbarsky-apple, 2023-08-10:
You say you hit compile errors, but you did not attach the log, so I have no idea what those errors were.  In general, for electrical measurement, we have no particular API, so you just write the attributes using the Accessors.h helpers.

aravindone, 2023-08-10:
Ok, I will close the issue . Will use this ticket alone https://github.com/project-chip/connectedhomeip/issues/28330

closed 2023-08-10 by aravindone