project-chip/connectedhomeip #28625: [BUG] [TE2] TC-IDM-10.1 Getting Unsupported Attribute After Reading Scenes Cluster's NameSupport  (Mandatory Attribute)

VaishaliAvhale, 2023-08-10:
### Reproduction steps

1. Commission a device
2. Run the below command
      ./chip-tool scenes read name-support 1 1
[Scenes-name-support-bug.log](https://github.com/project-chip/connectedhomeip/files/12313085/Scenes-name-support-bug.log)

      

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

core

bzbarsky-apple, 2023-08-10:
@VaishaliAvhale What example app are you using here, exactly?  Because this works fine for me with all-clusters-app from SHA d7414cc29442b3f29f9b978a67421a138f9c8861, but it has been supported for a while....

VaishaliAvhale, 2023-08-14:
It's my bad it's our code-side issue