project-chip/connectedhomeip #28630: [BUG] AddStatus error code is often ignored

andy31415, 2023-08-10:
### Reproduction steps

Examples from `rg -- '->AddStatus'`:

```
...
examples/ota-provider-app/ota-provider-common/OTAProviderExample.cpp:                commandObj->AddStatus(commandPath, Status::Failure);
examples/ota-provider-app/ota-provider-common/OTAProviderExample.cpp:    commandObj->AddStatus(commandPath, Status::Success);
src/controller/tests/data_model/TestCommands.cpp:            apCommandObj->AddStatusAndLogIfFailure(aCommandPath, Protocols::InteractionModel::Status::Failure,
src/controller/tests/data_model/TestCommands.cpp:            apCommandObj->AddStatus(aCommandPath, Protocols::InteractionModel::Status::Success);
src/controller/tests/data_model/TestCommands.cpp:                apCommandObj->AddStatusAndLogIfFailure(aCommandPath, Protocols::InteractionModel::Status::Succes
...
src/app/clusters/channel-server/channel-server.cpp:        command->AddStatus(commandPath, Status::Failure);
src/app/clusters/channel-server/channel-server.cpp:    command->AddStatus(commandPath, status);
src/app/clusters/channel-server/channel-server.cpp:    command->AddStatus(commandPath, status);
src/app/clusters/identify-server/identify-server.cpp:    commandObj->AddStatus(commandPath,
src/app/clusters/identify-server/identify-server.cpp:    commandObj->AddStatus(commandPath, Status::Success);
...
```

A large percentage of AddStatus commands are never actually checked for return codes.

This has two sideffects:
  - errors are never seen in logs
  - we have no sane way to handle AddStatus failures (would we return success without an error code?)

We likely would want to VerifyOrDie unless we are able to handle the error in some way.
Also need to review usage of `AddStatusAndLogIfFailure` since that both logs failure statuses (seems to be the intended functionality) but also adds a log of failed statuses (VerifyOrDie seems more correct though as we cannot really handle the failure).

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

ba490f6447d9fdc2aa8e6e20748b63bd7718d4f9

### Platform

core

bzbarsky-apple, 2023-08-11:
Reopening, since https://github.com/project-chip/connectedhomeip/pull/28634 was reverted....

andy31415, 2023-08-22:
Revert seems to have happened in #28661 due to OpenIOT perma-fail.

At the same time OpenIOT seems quite flaky in general (we had to disable several tests), so I am somewhat unclear on what is happening here.

PR #28634 (2023-08-10): Make AddStatus generally VerifyOrDie and have centralized logging
- Make AddStatus have integrated logging on failure and VerifyOrDie on failure
- Add `FailableAddStatus` when we think we can handle or forward the underlying CHIP_ERROR

Fixes #28630
PR #28800 (2023-08-22): Re-create "Make AddStatus generally VerifyOrDie and have centralized …
…logging".

This reverts the previous revert.

Specifically re-adds #28634 (reverting #28661).

Will check OpenIOT CI status.

Fixes #28630
closed 2023-08-22 by mergify[bot]