project-chip/connectedhomeip #28635: [BUG]

aravindone, 2023-08-10:
### Reproduction steps

im facing error will start the building of the esp32 examples.
steps:
1) activate the esp-idf v5.1 environment.
2) activate the  connectedhomeip environment.
3)move to examples/all-clusters-app/esp32.
4)idfy.py set-target esp32 (problem - stuck ).

[set-target esp32_error.TXT](https://github.com/project-chip/connectedhomeip/files/12314661/set-target.esp32_error.TXT)

does any python dependencies im missing for build ?

please let me know .. 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

7353fd6cab156b2f5cf5ca432f005cf33312a1fe 

### Platform

esp32

### Anything else?

@shubhamdp  @bzbarsky-apple  please comment.

shubhamdp, 2023-08-10:
https://github.com/project-chip/connectedhomeip/issues/27653#issuecomment-1635375210

bzbarsky-apple, 2023-08-10:
@aravindone Please stop pinging me on all esp32 issues.

aravindone, 2023-08-10:
Ok @bzbarsky-apple

aravindone, 2023-08-11:
from https://github.com/project-chip/connectedhomeip/issues/27653#issuecomment-1635375210 

is working fine im closing this ticket thanks.

@shubhamdp

closed 2023-08-11 by aravindone