project-chip/connectedhomeip #28648: [BUG] [chip-tool] Chip Tool may enter a dead loop when executing readEvent

ericzijian1994, 2023-08-11:
### Reproduction steps

This issue came up in TC-SMOKECO and I modified TestEvents.yaml to make it easier to find the issue.

```diff
diff --git a/src/app/tests/suites/TestEvents.yaml b/src/app/tests/suites/TestEvents.yaml
index 637268ecdc..97deb72493 100644
--- a/src/app/tests/suites/TestEvents.yaml
+++ b/src/app/tests/suites/TestEvents.yaml
@@ -89,6 +89,13 @@ tests:
               - name: "value"
                 value: eventNumber + 1
 
+    - label: "Read the event back (will fail)"
+      command: "readEvent"
+      event: "TestEvent"
+      response:
+          - values:
+                - value: { arg1: 1, arg2: 2, arg3: true }
+
     - label: "Read the event back"
       command: "readEvent"
       event: "TestEvent"
@@ -98,6 +105,17 @@ tests:
           - values:
                 - value: { arg1: 3, arg2: 1, arg3: false }
 
+    - label: "Read the event back (Will go into loop)"
+      command: "readEvent"
+      event: "TestEvent"
+      response:
+          - values:
+                - value: { arg1: 1, arg2: 2, arg3: true }
+          - values:
+                - value: { arg1: 3, arg2: 1, arg3: false }
+          - values:
+                - value: { arg1: 3, arg2: 1, arg3: false }
+
     - label: "Subscribe to the event"
       command: "subscribeEvent"
       event: "TestEvent"
```

I added two steps to the yaml where the response doesn't match what it actually is.

Run chip-tool: `./out/chip-tool/chip-tool tests TestEvents --nodeId 1 --continueOnFailure true > log.txt`

Return error: 

```text
CHIP:TOO:  ***** Step Failure: zzz_generated/chip-tool/zap-generated/test/Commands.h:89909: CHIP Error 0x0000002F: Invalid argument

...

CHIP:TOO:  ***** Step Failure: TestEvents

CHIP:TOO:  ***** Step Failure: TestEvents

CHIP:TOO:  ***** Step Failure: TestEvents

CHIP:TOO:  ***** Step Failure: TestEvents

... (go into loop)
```

Full log: 
[log.txt](https://github.com/project-chip/connectedhomeip/files/12318234/log.txt)

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

a98bc64856aa161197e7dc7c1ffbdcc43323eda3

### Platform

other

bzbarsky-apple, 2023-08-11:
@ericzijian1994 Is this also an issue with the Python chip-tool runner?  Or just with the codegen bits that we are trying to remove?

ericzijian1994, 2023-08-11:
No issues with Python chip-tool.
[python_chiptool_log.txt](https://github.com/project-chip/connectedhomeip/files/12318827/python_chiptool_log.txt)

bzbarsky-apple, 2023-08-11:
@ericzijian1994 Thank you for checking that!  Then we should just move forward with removing the codegen bits....

yunhanw-google, 2023-08-15:
@vivien-apple could you take a look at this and removing the codegen bits? thanks

closed 2023-08-24 by ericzijian1994