project-chip/connectedhomeip #28652: [BUG] CRASH: JNI DETECTED ERROR IN APPLICATION: JNI ERROR - ReleaseStringUTFChars

mukesh-simform, 2023-08-11:
### Reproduction steps

1. Discover available nodes in the network using `discoverCommissionableNodes`
2. Get Discovered Node using `getDiscoveredDevice`
3. Pair the device with the IP Address using `pairDeviceWithAddress`
4. ESP32 chip will be commissioned and shows proper logs but the application using SDK crashes immediately 

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

8f66f42

### Platform

android, esp32

### Platform Version(s)

v1.1.0.1

### Anything else?

The app is gettings crashed every time when tries to commission over an IP address using  `pairDeviceWithAddress` method

```
A  java_vm_ext.cc:594] JNI DETECTED ERROR IN APPLICATION: JNI ERROR (app bug): jstring is an invalid global reference: 0x418e (stale reference with serial number 3 v. current 6)

java_vm_ext.cc:594]     in call to ReleaseStringUTFChars
```

**Implementation**

[image]

**ESP32 Logs**

[image]

**Crash Logs Android**

[image]

yunhanw-google, 2023-08-15:
@joonhaengHeo could you help to take a look at this issue? thanks

pengxuetian, 2023-08-18:
[image]
@yunhanw-google Making these changes will fix the bug!

joonhaengHeo, 2023-08-21:
@pengxuetian 
I understood what the problem was. It would be nice if you could post a fix for this issue.

PR #28853 (2023-08-25): [Android] Fix SSID, Password String release crash
Fix #28652 

For avoiding this crash issue, I added null setting code after releasing jni string.
closed 2023-08-25 by mergify[bot]