project-chip/connectedhomeip #28658: [BUG] All clusters and all clusters minimal don't have an identify or scenes cluster on the onoff device type

cecille, 2023-08-11:
### Reproduction steps

see https://github.com/project-chip/connectedhomeip/pull/28647 - this gets added by zap, but probably won't function correctly.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

7353fd6cab156b2f5cf5ca432f005cf33312a1fe

### Platform

core

lpbeliveau-silabs, 2023-08-25:
The scenes cluster's support has been postponed to matter 1.3 so we can remove it from the all clusters app. It currently not spec compliant.

andy31415, 2023-08-29:
Currently it seems we do not fail all-cluster certification, so this may not be a cert blocker after all. Removing the tag, however @cjandhyala please add back with notes if this is indeed a problem.

cecille, 2023-08-30:
This isn't failing cert because of a test gap - device vs cluster testing. It may not be a current cert blocker, but please leave open

cecille, 2023-08-31:
You can untag as cert blocker though

jvmahon, 2023-10-11:
Shouldn't all-clusters apps be considered non-conforming since it adds clusters to endpoint types which don't have the clusters as either Mandatory or Optional?  As a longer-term update, I think it would be more helpful for example purposes if this could be an "All Endpoints" app with strictly conforming endpoints and complete implementations of those endpoints.