project-chip/connectedhomeip #28671: [BUG] Error 773 on commission

jonsmirl, 2023-08-12:
### Reproduction steps

Using off-the-shelf Nanoleaf A19 bulbs.
Commission with Google's Android sample app.
I also rebuilt the Android libs off from recent CHIP with no change.

Can someone show me where error 773 (0x305) can be generated in the CHIP code at the point shown in the log? 
If I can locate the point where the error code is generated I can debug more.

Error is line 573
[773.log](https://github.com/project-chip/connectedhomeip/files/12326072/773.log)

The Nanoleaf bulbs are on Thread so this maybe thread related.

### Bug prevalence

2-3 times a week

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

android

### Anything else?

This problem is intermittent and unpredictable. I can commission four identical bulbs, three will work, one will fail.

bzbarsky-apple, 2023-08-12:
If that 0x305 is a CHIP_ERROR value (which is not obvious from the logging), that would be an SDK error, SdkPart == 3 = kASN1, and then 05 is:
```
#define ASN1_ERROR_INVALID_ENCODING     CHIP_ASN1_ERROR(0x05)
```

jonsmirl, 2023-08-12:
@bzbarsky-apple Check out line 247. That looks like subscription data. Is that data coming from the node we are commissioning?

My app is listening to MDNS for commissioned nodes. As soon as it sees one it subscribes to it.

Does CHIP turn on the MDNS announcement before it sends OnCommissioningComplete?  It is possible my subscription got in front of OnCommissioningComplete on this slow Thread network?  Also notice there are a lot of retransmissions right before the error at 573.

I am counting on seeing the MDNS announcement as a signal it is safe to access the node.

jonsmirl, 2023-08-18:
I believe I found the problem. I was using random numbers for my certificate serial numbers. Sometimes those numbers were negative. OpenSSL does not consider this an error and let me generate the certificates.  However, mbed SSL does flag it as an error. If you go read the x509 spec, the spec says they have to be positive.  So 773, 0x305 was the real error. This also explains why it would work on a retry. 50/50 the serial number was negative.

What was very confusing was how this error did not show up until the onCommissioningComplete() command

closed 2023-08-18 by jonsmirl