project-chip/connectedhomeip #28676: [BUG] NXP K32W041AM platform compilation is broken

fuxiaoming-lumi, 2023-08-12:
### Reproduction steps

After https://github.com/project-chip/connectedhomeip/commit/0ecb4e163e16325868fbbbe8a4122968fcecab59 is merged, NXP K32W041AM platform compilation is broken.

1. Some configuration is hardcoded into the K32W061 platform. Should use **${device}**.
https://github.com/project-chip/connectedhomeip/blob/master/third_party/nxp/k32w0_sdk/k32w0_sdk.gni#L542-L545
https://github.com/project-chip/connectedhomeip/blob/master/third_party/nxp/k32w0_sdk/k32w0_sdk.gni#L207

2. The downloaded SDK is from https://mcuxpresso.nxp.com/ and the directory configuration does not match. For example，utilities are not in the root directory of the downloaded SDK.
https://github.com/project-chip/connectedhomeip/blob/master/third_party/nxp/k32w0_sdk/k32w0_sdk.gni#L602
```
SDK is from https://mcuxpresso.nxp.com/
.
├── boards
├── CMSIS
├── components
├── devices
├── docs
├── K32W041AMDK6_manifest_v3_4.xml
├── K32W041AMDK6_manifest_v3_5.xml
├── middleware
├── rtos
├── SW-Content-Register.txt
└── tools
```

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

1309e7d51f434d1954ab244562a8db6a01e87662

### Platform

k32w

### Platform Version(s)

N/A

marius-alex-tache, 2023-08-23:
The K32W0x SDK was moved to github, so we are now using `west` tool to update it directly in the repo: `third_party/nxp/k32w0_sdk/repo/core`.

From what I know, the packages for `K32W41A/AM` are generated based on the `K32W061` code. Unfortunately, I cannot change the paths to use `{device}` because of the current structure of the github SDK (I can signal the issue further).

Related to the second bullet, the user no longer needs to download the SDK. Instead, the user should use the above approach, as per the README file: https://github.com/project-chip/connectedhomeip/tree/master/examples/contact-sensor-app/nxp/k32w/k32w0#building.

fuxiaoming-lumi, 2023-08-23:
How about to reserve both ways to integrate K32W0 SDK?
Because it seems that there is consideration to configure the SDK path from the outside.
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/nxp/k32w/k32w0/args.gni#L24

marius-alex-tache, 2023-08-30:
I have raised the issue internally. I will let you know once there's a resolution. Thank you!

closed 2024-01-31 by fuxiaoming-lumi