project-chip/connectedhomeip #28677: [BUG]

gen-zi, 2023-08-13:
### Reproduction steps

two thread asynchronous involve ChipDeviceController.getConnectedDevicePointer to connect a device or one thread involve it frequently (maybe 20-50ms), that i got "Fatal signal 6"

there is two thread involve in the same time log: 
D/CSM: FindOrEstablishSession: PeerId = [1:0000000000000006]
D/CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
D/DIS: Found an existing secure session to [1:0000000000000006]!
D/DIS: OperationalSessionSetup[1:0000000000000006]: State change 1 --> 5
D/CSM: FindOrEstablishSession: PeerId = [1:0000000000000006]
D/DMG: SendReadRequest ReadClient[0x7c4d4a9450]: Sending Read Request
D/EM: <<< [E:51287i S:57535 M:20761014] (S) Msg TX to 1:0000000000000006 [98D9] --- Type 0001:02 (IM:ReadRequest)
D/IN: (S) Sending msg 20761014 on secure session with LSID: 57535
D/DMG: MoveToState ReadClient[0x7c4d4a9450]: Moving to [AwaitingIn]
D/DMG: SendReadRequest ReadClient[0x7c4d4afd50]: Sending Read Request
D/EM: <<< [E:51288i S:57535 M:20761015] (S) Msg TX to 1:0000000000000006 [98D9] --- Type 0001:02 (IM:ReadRequest)
D/IN: (S) Sending msg 20761015 on secure session with LSID: 57535
D/DMG: MoveToState ReadClient[0x7c4d4afd50]: Moving to [AwaitingIn]
A/libc: Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 23729 (Thread-36), pid 23471 

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

NA

### Platform

android

yunhanw-google, 2024-01-14:
@gen-zi recently we largely improve the stability on android chip-tool, could you retry this with latest code? If this is still problematic, we would take the further investigation on it, feel free to reach me at slack if you are in slack as well, thanks

yunhanw-google, 2024-02-22:
I cannot reproduce this issue using latest code, please retest using latest code, and feel free to reopen this one, thanks

closed 2024-02-22 by yunhanw-google