project-chip/connectedhomeip #28678: [BUG]

ImranR98, 2023-08-13:
### Reproduction steps

I'm trying to build the `chip-tool` for Linux (Fedora 38) with the following command run in the `connectedhomeip` repo:
```
./scripts/examples/gn_build_example.sh examples/chip-tool ../chip-tool
```

The build fails with the following error:
```
... <previous output not shown> ...
Done. Made 271 targets from 229 files in 125ms
+ ninja -C ../chip-tool
ninja: Entering directory `../chip-tool'
[1/31] c++ obj/third_party/connectedhomeip/third_party/perfetto/repo/sdk/sdk.perfetto.cc.o
FAILED: obj/third_party/connectedhomeip/third_party/perfetto/repo/sdk/sdk.perfetto.cc.o 
g++ -MMD -MF obj/third_party/connectedhomeip/third_party/perfetto/repo/sdk/sdk.perfetto.cc.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -Wno-shadow -Wno-format-nonliteral -std=gnu++17 -fno-rtti -Wnon-virtual-dtor  -I../connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/perfetto/repo/sdk -c ../connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/perfetto/repo/sdk/perfetto.cc -o obj/third_party/connectedhomeip/third_party/perfetto/repo/sdk/sdk.perfetto.cc.o
../connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/perfetto/repo/sdk/perfetto.cc: In member function ‘virtual void perfetto::ProducerIPCClientImpl::OnConnect()’:
../connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/perfetto/repo/sdk/perfetto.cc:78377:19: error: redundant move in initialization [-Werror=redundant-move]
78377 |     Sync(std::move(pending_sync));
      |          ~~~~~~~~~^~~~~~~~~~~~~~
../connectedhomeip/examples/chip-tool/third_party/connectedhomeip/third_party/perfetto/repo/sdk/perfetto.cc:78377:19: note: remove ‘std::move’ call
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

I've been following this guide and have done all the prerequisite steps as far as I can see: https://project-chip.github.io/connectedhomeip-doc/guides/chip_tool_guide.html

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

b0b0d58

### Platform

other

matt7aylor, 2023-08-18:
I encountered the same issue (on Arch Linux). Adding `-Wno-redundant-move` to the `CFLAGS` for `perfetto` did allow `chip-tool` to build (see below). I don't have knowledge/familiarity with codebase to dig any deeper into any underlying reason for the warning.

```diff
--- a/third_party/perfetto/BUILD.gn
+++ b/third_party/perfetto/BUILD.gn
@@ -18,7 +18,7 @@ import("//build_overrides/chip.gni")
 config("sdk_config") {
   include_dirs = [ "repo/sdk" ]
 
-  cflags = [ "-Wno-format-nonliteral" ]
+  cflags = [ "-Wno-format-nonliteral", "-Wno-redundant-move" ]
 }
 
 config("sdk_private_config") {
 ```

andy31415, 2023-08-28:
This is a compiler version thing.

Apparently #28842 updated this as a sideffect (and into private config) ...l could you try with that change?

andy31415, 2023-08-28:
From what I can tell, old gcc compilers required the move, new compilers complain that the move is there in the first place because it is redundant.

In any case, I believe this is fixed on ToT.

closed 2023-08-28 by andy31415