project-chip/connectedhomeip #28680: [BUG] Descriptor reporting with subscription

jonsmirl, 2023-08-14:
### Reproduction steps

Is this correct behavior? I have a wild card subscription and the Descriptor attributes are coming back in pieces.
Do other attributes report in pieces? If so, how can you differentiate a piece from a change in the attribute?

These are the events I am getting from the subscription API.
~~~
2023-08-14 01:26:28.250  3036-3121  DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Server ab0de00000000004 [29, 31, 40, 48, 49, 51, 60, 62, 63, 54, 4294179841]
2023-08-14 01:26:28.251  3036-3121  DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Client ab0de00000000004 []
2023-08-14 01:26:28.258  3036-3121  DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Server ab0de00000000004 [29, 3, 4, 5, 80, 30, 6, 8, 768, 59, 64, 65]
2023-08-14 01:26:28.259  3036-3121  DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Client ab0de00000000004 [6, 8, 768]
2023-08-14 01:26:28.261  3036-3121  DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Server ab0de00000000004 [29]
2023-08-14 01:26:28.262  3036-3121  DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Client ab0de00000000004 []
~~~

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

core

jonsmirl, 2023-08-17:
So how do I differentiate a single chuncked response from multiple responses?

bzbarsky-apple, 2023-08-28:
>  There needs to be an indicator in the response API to tell my code to combine or replace.

There is.  The attribute path tells you what's going on.  See the mListOp member of ConcreteDataAttributePath.

jonsmirl, 2023-08-29:
@bzbarsky-apple   Can you please reopen, I don't have the ability.

As far as I can see the Java API is not exposing mListOp.

https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/AndroidCallbacks.cpp#L254

ReportCallback::OnAttributeData receives the ConcreteDataAttributePath. When it creates the Java AttributeState object it does not include the mListOp status telling you to append, replace, etc.

jonsmirl, 2023-08-30:
@yunhanw-google Shouldn't this API be coalescing the chunking into a single value and not sending the piece into the user app?

yunhanw-google, 2023-08-30:
@jonsmirl Android's OnReportCallback use ClusterStateCache, and ClusterStateCache use  bufferedReadCallback
https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/AndroidCallbacks.h#L85
https://github.com/project-chip/connectedhomeip/blob/master/src/app/ClusterStateCache.h#L659
The chunked list has been coalesced into a single value in https://github.com/project-chip/connectedhomeip/blob/master/src/app/BufferedReadCallback.cpp#L84

Therefore when using Java/Jni's onReport, https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/AndroidCallbacks.cpp#L219
You should get a complete view for the NodeState, where you can retrieve each attribute as a whole under endpoint/cluster/attribute with tlv blob or json format.

jonsmirl, 2023-08-30:
So attributeState.getValue where the 'value' is returned as Java CHIPStrucs is basically unusable? I have to rework everything to use JSON or TLV directly? That is giant pain because I round trip these values. I modify the structs and then write them back to the CHIP attributes. And the CHIP API requires the CHIPStrucs.  I need to go explore Kotlin json support. Maybe the Kotlin json library can convert these back into CHIPStrucs. 

Does this same problem occur on a direct read of the attributes?

jonsmirl, 2023-08-30:
This is a total mess for user apps. Now I have to add code to convert every possible JSON object back into ChipStructs. That is the kind of code zap should be generating.  This API should be returning coalesced ChipStructs so that they can be round-tripped back into the CHIP API.

yunhanw-google, 2023-08-30:
You can still use attributeState.getValue with "USE_JAVA_TLV_ENCODE_DECODE" https://github.com/project-chip/connectedhomeip/blob/990bceaed0c3fd2af38561d4950f8a5e760c5d31/src/controller/java/BUILD.gn#L72. We try to avoid to generate the huge amount of generated IM code with USE_JAVA_TLV_ENCODE_DECODE, we could introduce another flag to control the only zap-generated/CHIPAttributeTLVValueDecoder.cpp

You could use JSON or Tlv.
We also provide the kotlin JSON/Tlv conversion support https://github.com/project-chip/connectedhomeip/tree/990bceaed0c3fd2af38561d4950f8a5e760c5d31/src/controller/java/src/chip/jsontlv

jonsmirl, 2023-08-30:
The JSON is identical to the ChipStruct, neither are coalesced.
~~~
2023-08-30 13:11:08.651 11409-11482 DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Server ab0de00000000019 [29, 31, 40, 48, 49, 51, 60, 62, 63, 54, 42]
2023-08-30 13:11:08.652 11409-11482 DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Server ab0de00000000019 {"value":[29,31,40,48,49,51,60,62,63,54,42]}
2023-08-30 13:11:08.653 11409-11482 DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Client ab0de00000000019 [41]
2023-08-30 13:11:08.655 11409-11482 DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Client ab0de00000000019 {"value":[41]}
2023-08-30 13:11:08.660 11409-11482 DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Server ab0de00000000019 [29, 3, 4, 5, 6, 8, 768]
2023-08-30 13:11:08.662 11409-11482 DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Server ab0de00000000019 {"value":[29,3,4,5,6,8,768]}
2023-08-30 13:11:08.663 11409-11482 DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Client ab0de00000000019 []
2023-08-30 13:11:08.664 11409-11482 DeviceServ...1$onReport com.lowpan.m2                        D  JDS Descriptor Client ab0de00000000019 {"value":[]}
~~~

jonsmirl, 2023-08-30:
> DeviceTypeList ServerList ClientList PartsList TagList
> 
> See these attributes, they are list, if it is too long to fit in a packet, the interaction model code would handle reports with chunks.
> 
> Chunking required when reading a large list or a large set of attributes. the minimal IPv6 MTU is 1280 bytes, and we use 1024 bytes as a safe upper bound for encoding interaction model payloads. Then it is not enough to fill the response in a single read response when there are a few fabrics on the device. The case for reading a large set of attributes is just the case for wildcard reading – we can only fit 30-40 AttributeDataIB-s in a single AttributeDataReport message, while reading all attributes in all clusters under on all endpoints will results in a report for hundreds of attributes, makes it critical for supporting chunking in report.

You said that earlier, but how does this work on Thread with an MTU of 127 bytes?  If only these (DeviceTypeList ServerList ClientList PartsList TagList) are impacted, I don't care I am just printing them out as an easy way to test.

But are ACLs going to get chunked on Thread? I can't tolerate partial ACL responses. Modifying the ACL has to be a read-modify-write operation.

yunhanw-google, 2023-08-30:
As I mentioned, from NodeState, you would see the attribute as a whole via json/tlv/ChipStruct, the underlying bufferReader in c++ layer has coalesced them together as a attribute if the list is oversized and sent via chunk

jonsmirl, 2023-08-30:
I printed it out above, the json is not coalesced.

Timber.d(
       "JDS Descriptor Server %x %s",
        deviceId,
        attributeState.json.toString()
}

yunhanw-google, 2023-08-30:
you may add additional log in https://github.com/project-chip/connectedhomeip/blob/master/src/app/BufferedReadCallback.cpp#L70, you may confirm whether the actual coalescing happens or not

closed 2023-08-30 by jonsmirl

yunhanw-google, 2023-08-30:
@jonsmirl https://github.com/project-chip/connectedhomeip/pull/28970
I just write one test, I create several attribute path, one list with serverList would be chunked, then I dump its json string, it shows the attribute is complete