project-chip/connectedhomeip #28682: [Feature] Support for Windows Platform in the Matter SDK

TheWalkingDead1024, 2023-08-14:
### Feature description

Background:
With the increasing adoption of the Matter SDK by developers globally, it is imperative to have compatibility across all major platforms. One significant platform that is widely used in both enterprise and consumer markets is Windows.

Feature Request:
I am proposing that the Matter SDK be extended to support the Windows platform. This would entail ensuring compatibility with various versions of Windows and potentially integrating with Windows-specific libraries or APIs.

Benefits:

Broader Adoption: By supporting Windows, the Matter SDK can be adopted by a wider range of developers, especially those focused on desktop applications and integrations.

Unified Development Experience: Developers often work on multiple platforms. Having Windows support ensures a seamless experience across different environments.

Enterprise Appeal: Many enterprises use Windows as their primary OS. By supporting Windows, Matter SDK becomes more appealing for enterprise-grade solutions.

Potential Challenges:

Compatibility: Ensuring the SDK is compatible with various versions of Windows might pose a challenge.

Integration: There might be Windows-specific libraries or APIs that the SDK needs to integrate with, which can be an added layer of complexity.

I sincerely hope the team considers this feature request, as it would not only enhance the usability of the SDK but also significantly increase its adoption rate.

### Platform

windows

TheWalkingDead1024, 2023-08-15:
I think this is a issue with Multiple Initializations of DeviceControllerFactory in Android

Single Initialization in iOS:
In iOS, the running status of DeviceControllerFactory is indirectly determined through the isRunning method. As a result, the init method of DeviceControllerFactory is called only once. Therefore, there is no need for DeviceControllerFactory to execute a shutdown.
[image]

Multiple Initializations in Android:
In contrast, in the Android environment, each time a new controller is generated, DeviceControllerFactory is initialized once again. This behavior is causing the “Device Controller Factory already initialized...” warning message to be displayed.
[image]

Potential Solutions:
To resolve this issue, one approach could be to add logic within the init method to prevent multiple initializations of DeviceControllerFactory.
Alternatively, executing a shutdown for the factory before initializing it again could be another solution.
[image]

However, we are not familiar with the underlying details of the Matter stack, and we are concerned that making such changes might introduce other unforeseen issues. Hence, we are raising this issue for your attention and action.
Request:
- We kindly request the team to look into this issue and make the necessary corrections to ensure that the DeviceControllerFactory is managed consistently across both iOS and Android platforms.

dotMorten, 2023-10-17:
I think this is a duplicate of https://github.com/project-chip/connectedhomeip/issues/24223

tomxue, 2025-01-16:
Same question. Is there any update?

richiemsft, 2026-09-15:
Give this a try:
I forked the repo https://github.com/richiemsft/matterforwindows/tree/windows-port and started working on native windows support.
Details on how to build: https://github.com/richiemsft/matterforwindows/blob/windows-port/docs/guides/windows.md