project-chip/connectedhomeip #28689: [Build] chip-tool build fails on Raspberry Pi 4 (Ubuntu OS)

Peugot206, 2023-08-14:
### Build issue(s)

Following this guide: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md for the Raspberry Pi 4, running Ubuntu 20.04 64-bit server.

After source scripts/activate.sh is completed succesfully I want to build the CHIP-tool using the following command:
`./scripts/examples/gn_build_example.sh examples/chip-tool ~/CHIP_TOOL_BUILD` as explained in the [CHIP-tool guide](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md#step-7-control-application-data-model-clusters).

After quite some time I get the following error: 
```
[P5737 12:02:15.292 storage.go:245 I] [fetch 3/4] Fetching:  63.6/405.8 MB ( 16%, 5.89 MB/s)
[P5737 12:02:15.793 storage.go:245 I] [fetch 3/4] Fetching:  66.7/405.8 MB ( 16%, 5.92 MB/s)
[P5737 12:02:16.294 storage.go:245 I] [fetch 3/4] Fetching:  68.9/405.8 MB ( 17%, 5.68 MB/s)
[P5737 [P5737 12:07:27.508 client.go:1913 E] [unzip 3/4] Failed to install fuchsia/third_party/rust/linux-arm64 in "pigweed/rust": failed to download after multiple attempts
Errors:
  failed to install fuchsia/third_party/clang/linux-arm64 in "pigweed": failed to download after multiple attempts
  failed to install fuchsia/third_party/sysroot/linux in "pigweed/clang_sysroot": write /home/userpi/connectedhomeip/.environment/cipd/packages/.cipd/pkgs/8/YuDp0GW7uRHE/usr/lib/python3.8/config-3.8-arm-linux-gnueabihf/libpython3.8-pic.a: no space left on device (and 4471 other errors)
  failed to install fuchsia/third_party/zap/linux-arm64 in "zap": write /home/userpi/connectedhomeip/.environment/cipd/packages/.cipd/pkgs/7/prABZHrcLuSr/resources/app.asar: no space left on device (and 114 other errors)
  failed to install fuchsia/third_party/rust/linux-arm64 in "pigweed/rust": failed to download after multiple attempts
/home/userpi/connectedhomeip/.environment/cipd/cipd ensure -ensure-file /home/userpi/connectedhomeip/.environment/cipd/packages.ensure -root /home/userpi/connectedhomeip/.environment/cipd/packages -log-level debug -json-output /home/userpi/connectedhomeip/.environment/cipd/packages.json -cache-dir /home/userpi/.cipd-cache-dir -max-threads 0
None
Traceback (most recent call last):
  File "/home/userpi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 932, in <module>
    sys.exit(main())
  File "/home/userpi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 924, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/userpi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 539, in setup
    result = step(spin)
  File "/home/userpi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 652, in cipd
    if not cipd_update.update(
  File "/home/userpi/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 382, in update
    subprocess.check_call(cmd, stdout=outs, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/userpi/connectedhomeip/.environment/cipd/cipd', 'ensure', '-ensure-file', '/home/userpi/connectedhomeip/.environment/cipd/packages.ensure', '-root', '/home/userpi/connectedhomeip/.environment/cipd/packages', '-log-level', 'debug', '-json-output', '/home/userpi/connectedhomeip/.environment/cipd/packages.json', '-cache-dir', '/home/userpi/.cipd-cache-dir', '-max-threads', '0']' returned non-zero exit status 1.
```

I hope anyone could help me out here, if you need any additional information, please let me know.

Thanks for any help.

### Platform

raspi

aravindone, 2023-08-15:
@Peugot206 

`Errors:
  failed to install fuchsia/third_party/clang/linux-arm64 in "pigweed": failed to download after multiple attempts
  failed to install fuchsia/third_party/sysroot/linux in "pigweed/clang_sysroot": write /home/userpi/connectedhomeip/.environment/cipd/packages/.cipd/pkgs/8/YuDp0GW7uRHE/usr/lib/python3.8/config-3.8-arm-linux-gnueabihf/libpython3.8-pic.a: no space left on device (and 4471 other errors)
  failed to install fuchsia/third_party/zap/linux-arm64 in "zap": write /home/userpi/connectedhomeip/.environment/cipd/packages/.cipd/pkgs/7/prABZHrcLuSr/resources/app.asar: no space left on device (and 114 other errors)
  failed to install fuchsia/third_party/rust/linux-arm64 in "pigweed/rust": failed to download after multiple attempts`

fromthe logs it shows that there is no space on device could you please check that , and try.

Peugot206, 2023-08-15:
You are right, the device is out of space. I have a 32GB SD on the RPI and somehow it is already full after installing all the necessary Matter software. Going to upgrade to a 64GB SD card.

@aravindone do you know if Matter builds on Raspberry Pi OS 64-bit? Since it has to be a 64-bit system, this is now available on the RPI OS.

aravindone, 2023-08-15:
@Peugot206 matter usage on raspberry pi using Ubuntu server OS is the best option  ### Raspberry Pi OS 64-bit###  wont build properly.

Peugot206, 2023-08-17:
> @Peugot206 matter usage on raspberry pi using Ubuntu server OS is the best option . Raspberry Pi OS 64-bit im not sure

Thanks @aravindone. I tried and it indeed does not work on Raspberry Pi OS 64-bit. I am now installing on Ubuntu Server 22.04 on a Raspberry Pi 4 (4 GB, 64 GB storage USB) and it keeps getting stuck on the command after a while: 

`./scripts/examples/gn_build_example.sh examples/chip-tool BUILD_PATH`

as explained [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md#building-the-chip-tool). I then have to reboot the Pi, run the command again and it seems to pick up where it stopped, but stops very quickly again.

Any clue why? Could anyone else try this on a Raspberry Pi 4 to see if he has any success. Am I also right that this installation is pretty huge and almost requires ~25+ GB? I think that should be mentioned in the build instructions at least.

aravindone, 2023-08-18:
>  it keeps getting stuck on the command after a while

@Peugot206 could you please share the log for analyzing problem. more over if your working on raspberry pi it will be little slow.

Peugot206, 2023-08-18:
> could you please share the log for analyzing problem. more over if your working on raspberry pi it will be little slow.

@aravindone hi, I don't really have the logs but had a screenshot from where it got stuck, this was with the command `./scripts/examples/gn_build_example.sh examples/chip-tool BUILD_PATH`: 

[image]

As you can see now it shows [0/192], if one file takes too long to build, it eats up all the RAM and eventually gets stuck. With enabling 3 GB of swap only then I could get through this build process. Some files will build almost instantly but some other (presumably big ones) take a long time. Luckily if it gets stuck it continues where it got stuck on, and doesn't restart all over again.

Is this normal? I thought many people used the Pi to run chip-tool on..

aravindone, 2023-08-22:
Hi @Peugot206 i would suggest to go for optimized approach raspberry pi image  like : https://github.com/SiliconLabs/matter/blob/release_1.1.0-1.1/docs/silabs/general/ARTIFACTS.md

this image contains the OTBR and chip-tool bult-in , because matter sdk will build from scratch which eventually end up in stuck like this.

but for matter commissioning i would recommend for ubuntu laptop for best practice.

Peugot206, 2023-08-24:
Thanks a lot @aravindone ! I will use this for future use cases. 

This does mean it won't be the latest version, right? And I can't just 'update' to the latest version, but it is more like a snapshot. If that's the case, it still shouldn't really be a problem for me.

bzbarsky-apple, 2023-08-28:
Initial report is solved with the "out of space" observation.  Other problems should go into separate issues.

closed 2023-08-28 by bzbarsky-apple