project-chip/connectedhomeip #28694: [BUG]  Memory Leak in ResolverProxy

CodeChronos928, 2023-08-14:
### Reproduction steps

I just finished triaging a memory leak in `ResolverProxy`.

This memory allocation is getting leaked ([Source code link](https://github.com/project-chip/connectedhomeip/blob/3c3e5f52ddb7b73bee4ca3c0d9f4584ddd98dced/src/lib/dnssd/ResolverProxy.h#L89)): 

```c++
    CHIP_ERROR Init(Inet::EndPointManager<Inet::UDPEndPoint> * udpEndPoint = nullptr) override
    {
        ReturnErrorOnFailure(chip::Dnssd::Resolver::Instance().Init(udpEndPoint));
        VerifyOrReturnError(mDelegate == nullptr, CHIP_ERROR_INCORRECT_STATE);
-->     mDelegate = chip::Platform::New<ResolverDelegateProxy>();   <-- // Allocated and never freed.

...
```

This was the call stack of the leaked allocation:

```c++
src/lib/support/CHIPMem-Malloc.cpp: chip::Platform::MemoryAlloc()
src/lib/support/CHIPMem.h: chip::Platform::New<>()
src/lib/dnssd/ResolverProxy.h: chip::Dnssd::ResolverProxy::Init()
src/lib/dnssd/Discovery_ImplPlatform.cpp: chip::Dnssd::DiscoveryImplPlatform::HandleDnssdInit()
src/platform/google/dnssd_manager.cc: chip::Dnssd::ChipDnssdInit()
src/lib/dnssd/Discovery_ImplPlatform.cpp: chip::Dnssd::DiscoveryImplPlatform::InitImpl()
src/lib/dnssd/Discovery_ImplPlatform.h: chip::Dnssd::DiscoveryImplPlatform::Init()
src/lib/dnssd/Discovery_ImplPlatform.h: chip::Dnssd::DiscoveryImplPlatform::Init()
src/controller/CHIPDeviceControllerFactory.cpp: chip::Controller::DeviceControllerFactory::InitSystemState()
src/controller/CHIPDeviceControllerFactory.cpp: chip::Controller::DeviceControllerFactory::Init()
```

Although everything shuts down properly, This call to `mDelegate->Release()` in `ResolverProxy::Shutdown` is not freeing mDelegate as expected, as its ref count is still greater than 1 ([Source code link](https://github.com/project-chip/connectedhomeip/blob/3c3e5f52ddb7b73bee4ca3c0d9f4584ddd98dced/src/lib/dnssd/ResolverProxy.h#L150)):

```c++
    void Shutdown() override
    {
        VerifyOrReturn(mDelegate != nullptr);
        mDelegate->SetOperationalDelegate(nullptr);
        mDelegate->SetCommissioningDelegate(nullptr);
-->     mDelegate->Release(); <-- // Does not free mDelegate (ref count is > 1)
        mDelegate = nullptr;
    }
```

The reason is that `ResolverProxy::ResolveNodeId` obtained an additional ref to it here, and never released it ([Source code link](https://github.com/project-chip/connectedhomeip/blob/3c3e5f52ddb7b73bee4ca3c0d9f4584ddd98dced/src/lib/dnssd/Discovery_ImplPlatform.cpp#L701)):

```c++
CHIP_ERROR ResolverProxy::ResolveNodeId(const PeerId & peerId)
{
    VerifyOrReturnError(mDelegate != nullptr, CHIP_ERROR_INCORRECT_STATE);

    ChipLogProgress(Discovery, "Resolving " ChipLogFormatX64 ":" ChipLogFormatX64 " ...",
                    ChipLogValueX64(peerId.GetCompressedFabricId()), ChipLogValueX64(peerId.GetNodeId()));

    DnssdService service;

    ReturnErrorOnFailure(MakeInstanceName(service.mName, sizeof(service.mName), peerId));
    Platform::CopyString(service.mType, kOperationalServiceName);
    service.mProtocol    = DnssdServiceProtocol::kDnssdProtocolTcp;
    service.mAddressType = Inet::IPAddressType::kAny;

--> mDelegate->Retain(); <-- // Increases the ref-count of mDelegate.

    CHIP_ERROR err = ChipDnssdResolve(&service, Inet::InterfaceId::Null(), HandleNodeIdResolve, mDelegate);
    if (err != CHIP_NO_ERROR)
    {
        mDelegate->Release();
    }
    return err;
}

```

In my specific use-case, it was expected to be released when `ResolverProxy::NodeIdResolutionNoLongerNeeded` is called, but it's clearly not ([Source code link](https://github.com/project-chip/connectedhomeip/blob/3c3e5f52ddb7b73bee4ca3c0d9f4584ddd98dced/src/lib/dnssd/Discovery_ImplPlatform.cpp#L711)):

```c++
void ResolverProxy::NodeIdResolutionNoLongerNeeded(const PeerId & peerId)
{
    char name[Common::kInstanceNameMaxLength + 1];
    ReturnOnFailure(MakeInstanceName(name, sizeof(name), peerId));

    ChipDnssdResolveNoLongerNeeded(name);
}
```

Actually, `DiscoveryImplPlatform` doesn't even propagate the call to `DiscoveryImplPlatform::NodeIdResolutionNoLongerNeeded` to the `ResolverProxy`, which I think it should be, and surrounding functions do this ([Source code link](https://github.com/project-chip/connectedhomeip/blob/3c3e5f52ddb7b73bee4ca3c0d9f4584ddd98dced/src/lib/dnssd/Discovery_ImplPlatform.cpp#L640)).

```c++
CHIP_ERROR DiscoveryImplPlatform::ResolveNodeId(const PeerId & peerId)
{
    ReturnErrorOnFailure(InitImpl());
--> return mResolverProxy.ResolveNodeId(peerId);  <-- // Calls ResolverProxy.
}

void DiscoveryImplPlatform::NodeIdResolutionNoLongerNeeded(const PeerId & peerId)
{
    char name[Common::kInstanceNameMaxLength + 1];
    ReturnOnFailure(MakeInstanceName(name, sizeof(name), peerId));

-->  ChipDnssdResolveNoLongerNeeded(name);  <-- // Does NOT call ResolverProxy.
}

CHIP_ERROR DiscoveryImplPlatform::DiscoverCommissionableNodes(DiscoveryFilter filter)
{
    ReturnErrorOnFailure(InitImpl());
--> return mResolverProxy.DiscoverCommissionableNodes(filter);  <-- // Calls ResolverProxy.
}
```

### Thoughts on a fix

We need to call `ResolverProxy::NodeIdResolutionNoLongerNeeded` from `DiscoveryImplPlatform::NodeIdResolutionNoLongerNeeded`, and `ResolverProxy::NodeIdResolutionNoLongerNeeded` should be calling `mDelegate->Release()` if the specified PeerId is holding a ref.

We'll need a set of PeerIds to keep track of how many times we've retained `mDelegate`, since we don't want to decrease the ref-count if someone calls `ResolverProxy::NodeIdResolutionNoLongerNeeded` with a PeerId that doesn't have an active reference to `mDelegate`.  That risks prematurely freeing `mDelegate` which would lead to a crash.

### Other concerns

It looks like there are a few other calls to `mDelegate->Retain()` in `Discovery_ImplPlatform.cpp` as well, and I don't see many calls to `mDelegate->Release()` so there may be other execution flows that leak the delegate as well.

### Bug prevalence

Whenever the AddressResolve logic is used, and we no longer need to resolve a node.

### GitHub hash of the SDK that was being used

af6cb3a6d7d4cd9a4eb38ec49687260bc5ec96e7

### Platform

core

### Anything else?

Although the hash I reported isn't the latest, I inspected the top-of-tree code, and the issue still prevails.  All my permalinks are from the top of tree source.

bzbarsky-apple, 2023-08-28:
The right fix here is to stop having ResolveNodeId go through a singleton resolver proxy (and in general to remove ResolveNodeId from ResolverProxy, since we are no longer using per-resolve resolver proxies for doing PeerId lookups).

andy31415, 2023-09-14:
I can reproduce the leak by stopping a test app while it is running resolution. Traces look like:

```
==1799042==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 112 byte(s) in 1 object(s) allocated from:
    #0 0x7f16f5eae7ff in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x559cfbdb7d92 in chip::Platform::MemoryAlloc(unsigned long) ../../src/lib/support/CHIPMem-Malloc.cpp:91
    #2 0x559cfbe223f2 in chip::Dnssd::MdnsAvahi::ResolveContext* chip::Platform::New<chip::Dnssd::MdnsAvahi::ResolveContext>() ../../src/lib/support/CHIPMem.h:145
    #3 0x559cfbe1c25a in chip::Dnssd::MdnsAvahi::Resolve(char const*, char const*, chip::Dnssd::DnssdServiceProtocol, chip::Inet::IPAddressType, chip::Inet::IPAddressType, chip::Inet::InterfaceId, void (*)(void*, chip::Dnssd::DnssdService*, chip::Span<chip::Inet::IPAddress> const&, chip::ChipError), void*) ../../src/platform/Linux/DnssdImpl.cpp:788
    #4 0x559cfbe1ead2 in chip::Dnssd::ChipDnssdResolve(chip::Dnssd::DnssdService*, chip::Inet::InterfaceId, void (*)(void*, chip::Dnssd::DnssdService*, chip::Span<chip::Inet::IPAddress> const&, chip::ChipError), void*) ../../src/platform/Linux/DnssdImpl.cpp:988
    #5 0x559cfbdcdbe5 in chip::Dnssd::ResolverProxy::ResolveNodeId(chip::PeerId const&) ../../src/lib/dnssd/Discovery_ImplPlatform.cpp:703
    #6 0x559cfbdcc97b in chip::Dnssd::DiscoveryImplPlatform::ResolveNodeId(chip::PeerId const&) (/usr/local/google/home/andreilitvin/devel/connectedhomeip/out/linux-x64-leak-checker-platform-mdns-asan/leak-checker+0x6e797b)
    #7 0x559cfbdb0a5f in chip::AddressResolve::Impl::Resolver::LookupNode(chip::AddressResolve::NodeLookupRequest const&, chip::AddressResolve::Impl::NodeLookupHandle&) ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:189
    #8 0x559cfba3c454 in (anonymous namespace)::leakcheck::SendRequest() (/usr/local/google/home/andreilitvin/devel/connectedhomeip/out/linux-x64-leak-checker-platform-mdns-asan/leak-checker+0x357454)
    #9 0x559cfba3c61d in (anonymous namespace)::leakcheck::PlatformEventHandler(chip::DeviceLayer::ChipDeviceEvent const*, long) (/usr/local/google/home/andreilitvin/devel/connectedhomeip/out/linux-x64-leak-checker-platform-mdns-asan/leak-checker+0x35761d)
    #10 0x559cfbde4ce6 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToApplication(chip::DeviceLayer::ChipDeviceEvent const*) ../../src/include/platform/internal/GenericPlatformManagerImpl.ipp:336
    #11 0x559cfbde49cf in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) ../../src/include/platform/internal/GenericPlatformManagerImpl.ipp:301
    #12 0x559cfbde2bda in chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) ../../src/include/platform/PlatformManager.h:503
    #13 0x559cfbde5b9d in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::ProcessDeviceEvents() ../../src/include/platform/internal/GenericPlatformManagerImpl_POSIX.ipp:182
    #14 0x559cfbde54fd in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() ../../src/include/platform/internal/GenericPlatformManagerImpl_POSIX.ipp:224
    #15 0x559cfba3b2f7 in chip::DeviceLayer::PlatformManager::RunEventLoop() ../../src/include/platform/PlatformManager.h:403
    #16 0x559cfba3c778 in (anonymous namespace)::Cmd_Check(int, char const**) (/usr/local/google/home/andreilitvin/devel/connectedhomeip/out/linux-x64-leak-checker-platform-mdns-asan/leak-checker+0x357778)
    #17 0x559cfba3d8eb in main (/usr/local/google/home/andreilitvin/devel/connectedhomeip/out/linux-x64-leak-checker-platform-mdns-asan/leak-checker+0x3588eb)
    #18 0x7f16f51666c9 in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

Indirect leak of 40 byte(s) in 1 object(s) allocated from:
    #0 0x7f16f5eae7ff in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x559cfbdb7d92 in chip::Platform::MemoryAlloc(unsigned long) ../../src/lib/support/CHIPMem-Malloc.cpp:91
    #2 0x559cfbdd05a2 in chip::Dnssd::ResolverDelegateProxy* chip::Platform::New<chip::Dnssd::ResolverDelegateProxy>() ../../src/lib/support/CHIPMem.h:145
    #3 0x559cfbdc2a7a in chip::Dnssd::ResolverProxy::Init(chip::Inet::EndPointManager<chip::Inet::UDPEndPoint>*) ../../src/lib/dnssd/ResolverProxy.h:89
    #4 0x559cfbdc7ae1 in chip::Dnssd::DiscoveryImplPlatform::HandleDnssdInit(void*, chip::ChipError) (/usr/local/google/home/andreilitvin/devel/connectedhomeip/out/linux-x64-leak-checker-platform-mdns-asan/leak-checker+0x6e2ae1)
    #5 0x559cfbe16ea9 in chip::Dnssd::MdnsAvahi::HandleClientState(AvahiClient*, AvahiClientState) (/usr/local/google/home/andreilitvin/devel/connectedhomeip/out/linux-x64-leak-checker-platform-mdns-asan/leak-checker+0x731ea9)
    #6 0x559cfbe16bfc in chip::Dnssd::MdnsAvahi::HandleClientState(AvahiClient*, AvahiClientState, void*) (/usr/local/google/home/andreilitvin/devel/connectedhomeip/out/linux-x64-leak-checker-platform-mdns-asan/leak-checker+0x731bfc)
    #7 0x7f16f682ab6f  (/lib/x86_64-linux-gnu/libavahi-client.so.3+0x4b6f)

SUMMARY: AddressSanitizer: 152 byte(s) leaked in 2 allocation(s).
```

andy31415, 2023-09-14:
To reproduce you can compile and run `./out/linux-x64-address-resolve-tool-platform-mdns-asan/address-resolve-tool node 1 2` and just ctrl-C out of it while resolution is running.

PR #29264 (2023-09-15): Stop using ResolverProxy for resolving node lookups
Fixes #28694 (memory leak because stopping resolution was not implemented).

We currently use AddressResolve to handle node resolution, so a separate singleton resolver proxy should not be needed for handling node resolution.

Actual changes:
  - remove implementation for ResolverProxy::ResolveNodeId
  - replace with inline code for PlatformDnsSd for resolverproxy. Made PlatformDnsSd maintain its own delegate for operational node discovery
  - Updated Avahi linux implementation to support "StopResolution" by kepping track of active resolution [... cut]
closed 2023-09-15 by mergify[bot]