project-chip/connectedhomeip #28700: missing and no known rule to make it

chuckrui, 2023-08-15:
When I build chip-tool according to the website, I found some errors, can you help me solve it?
[https://github.com/nrfconnect/sdk-connectedhomeip/blob/9e6386ca7506b04beccba7ec7054054b48ae4d7e/docs/guides/BUILDING.md#prerequisites](url)

[image]
Synchronizing submodules does not have any problems, but why does this synchronizing submodules display missing files?
[image]

chuckrui, 2023-08-15:
run gn_build.sh log
[gn_build.txt](https://github.com/project-chip/connectedhomeip/files/12341938/gn_build.txt)

aravindone, 2023-08-15:
@chuckrui  use the below command after activateing matter sdk to build chip tool , follow steps from url

`./scripts/examples/gn_build_example.sh examples/chip-tool BUILD_PATH`

[https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md](url)

closed 2023-08-17 by chuckrui