project-chip/connectedhomeip #28702: [CERT-TEST-FAILURE]

SamCullen-Element, 2023-08-15:
### Feature Area

Other

### Test Case

N/A

### Reproduction steps

Python scripts are not printing the outgoing frames when sending commands, we only see the incoming responds printed out. visibility of the outgoing frames was a request made from the SVE1 of Matter for reviewer confidence that the step is sending what we need it to send. Please re-enable this.

### Bug prevalence

Every Python script

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

other

andy31415, 2023-08-21:
https://github.com/project-chip/connectedhomeip/pull/28179 added python tracing bits, but using the updated trace decoder. This however should make reading things simpler than the `trace_decode`. Unsure if this was plumbed through test execution though.

andy31415, 2023-08-21:
@SamCullen-Element could you give me an example on a test you run where you need this info?

I am trying to make sure that I make it available in a way that is useful. A command as well as actual usage/requirements would help since the tracing data is slightly different between chip-tool and the tracing that we have out of the box in python.

cecille, 2023-08-21:
Is this a cert blocker or a feature request?

andy31415, 2023-08-22:
I merged #28785 which should enable `--trace-to json:<path>` and `--trace-to json:log` output for both outgoing and incoming data frames for python ... finding out where to place these in command line arguments is what is needed though.

Could you give me a concrete example to try out (for example a color control test execution) so that I can provide what arguments have to be added to the test run (or if it does not work to actually fix it)?

closed 2023-08-24 by mergify[bot]