project-chip/connectedhomeip #28723: [1.1] pairing fail (Error 0x00000032:timeout) with chip-tool pairing ble-thread command

ray24c, 2023-08-17:
### Reproduction steps

1. pairing a matter end device to the OTBR
2. issue command: sudo ./chip-tool pairing ble-thread 1301 \
hex:0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00\
beef0000051012d73fd1db535e043647e2371714df1a030a4f70656e5468726561640102d0360410c6\
ca22b09c62d22718bb3c05b32926ae0c0402a0f7f8 20202021 3840
3. I receive a fail as attach
[otbr_fail_log.txt](https://github.com/project-chip/connectedhomeip/files/12364836/otbr_fail_log.txt)

### Bug prevalence

when it happen, it will keep happen

### GitHub hash of the SDK that was being used

2023.03

### Platform

raspi

### Type

Core SDK Interopability Issue

gokul-bhalodiya-7, 2023-08-17:
As per the logs, OTBR failed at "ThreadNetworkEnable" commissioning stage.

You should confirm that the thread network is formed and active on the border router side using "ot-ctl state" command. 

You should check that the openthread dataset credentials on border router side is the same as the openthread dataset credentials given as argument to chip-tool while commissioning end-device.

You can display openthread dataset value by running command "ot-ctl dataset active -x" at thread border router side and give this value to the chip-tool as dataset value argument.

Having same network configuration settings should resolve the issue.

Peugot206, 2023-08-18:
Hi I have the same issue @gokul-bhalodiya-7, 

My RPI running OTBR returns `sudo ot-ctl state` leader. My other Thread/Matter device returns "otcli state" with "child" and sometimes "router". I made sure that the `dataset active -x` hex value is the same for the OTBR as the other Thread/Matter device, using `dataset active -x` on one device and setting it on the other device using `dataset set active [HEX]`.

For me it fails here:
```
[1692344892.013863][2164:2166] CHIP:CTL: Error on commissioning step 'ThreadNetworkSetup': 'src/controller/CHIPDeviceController.cpp:2268: CHIP Error 0x000000AC: Internal error'
[1692344892.013919][2164:2166] CHIP:CTL: Failed to perform commissioning step 21
[1692344892.013984][2164:2166] CHIP:CTL: Going from commissioning step 'ThreadNetworkSetup' with lastErr = 'src/controller/CHIPDeviceController.cpp:2268: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'
[1692344892.014061][2164:2166] CHIP:CTL: Performing next commissioning step 'Cleanup' with completion status = 'src/controller/CHIPDeviceController.cpp:2268: CHIP Error 0x000000AC: Internal error'
[1692344892.014122][2164:2166] CHIP:CTL: Successfully finished commissioning step 'Cleanup'
[1692344892.014195][2164:2166] CHIP:TOO: Device commissioning Failure: src/controller/CHIPDeviceController.cpp:2268: CHIP Error 0x000000AC: Internal error
``` 

And finally at the end with:
```
[1692344946.892759][2181:2181] CHIP:TOO: Run command failure: src/platform/Linux/bluez/ChipDeviceScanner.cpp:173: CHIP Error 0x00000032: Timeout
```

I made sure the device ble's adv is enabled. I can confirm that part works, because when I disable it I get immediately a different error.

gokul-bhalodiya-7, 2023-08-27:
It looks like that matter controller is **not able to connect to matter end-device in the thread network** so after that it gives timeout error.

You can try the below mentioned 2 scenarios to debug what's the problem:
1. Try pairing your matter controller with another matter end-device having thread support.
2. If you have any thread device which has the access to the OpenThread CLI then give thread credentials of the Controller to this end-device using the OpenThread CLI and debug whether the end-device is able to join the thread network or not.