project-chip/connectedhomeip #28731: [1.2] [BUG] Invalid YAML tests

plauric, 2023-08-17:
### Reproduction steps

This issue was introduced by the changes from this PR:
https://github.com/project-chip/connectedhomeip/pull/28656

Apparently the PR above reverted the YAML scripts to the version used in Matter 1.1, in order to address several issues, including https://github.com/CHIP-Specifications/chip-test-scripts/issues/601: "Since event list has moved out of 1.2, multiple test scripts needs to be revert back to 1.1 version prior to SVE."

The problem is that for many scripts reverting to the version from Matter 1.1 removed a lot of other changes, in addition to the event list one. 

I didn't investigate the scope of this unwanted change, but at least the following files are affected: Test_TC_RVCRUNM_1_1.yaml, Test_TC_RVCCLEANM_1_1.yaml, Test_TC_RVCOPSTATE_1_1.yaml, Test_TC_OPSTATE_1_1.yaml

Please revert the changes made by the PR.

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

76848fe  

### Platform

core

### Type

Test Improvement

raju-apple, 2023-08-17:
@manjunath-grl can you double check to make sure the version of YAML that you have modified is only w.r.t disabling the EventList steps but rest of the Yaml is same as it was before the change. We don't want to revert it to 1.1 version as lot of other changes would have been added to the yaml as well.

manjunath-grl, 2023-08-22:
@plauric We are not sure which change you refer as "revert". I don't recall reverting any yaml to Matter 1.1. Only change we have done to those scripts you mentioned are,
1. Adding Step numbers to the label
2. Adding prompt steps of AttributeList, GeneratedCommandList, AcceptedCommandList and EventList. 
3. For the EventList prompts we have also added `PICS_EVENT_LIST_ENABLED` flag. 

Please correct me if I am wrong. 

CC: @raju-apple

manjunath-grl, 2023-08-24:
@plauric Please check the diff link below which shows the difference between 1.1 and current version.
Diff link: https://gist.github.com/manjunath-grl/8b06fc9651ba8a5c3a233dd4b889a252/revisions?diff=unified

plauric, 2023-08-31:
@manjunath-grl It looks like this issue was addressed by the changes from this PR so I'll close it: https://github.com/project-chip/connectedhomeip/pull/28944/files

closed 2023-08-31 by plauric