project-chip/connectedhomeip #28738: [BUG] chip tool times out while commissioning both custom device and/or NRF52840DK

adithyakirank, 2023-08-17:
### Reproduction steps

Hey, I am using an ESP32-S3 based thread border router (the reference design) and have tried to commission my custom matter device and the NRF52840DK with the matter light bulb example. My commissioning command looks like this:

For my custom device:

```
./chip-tool pairing ble-thread 1 hex:0E080000000000010000000300000F35060004001FFFE00208D0AD1111D0AD22220708FD993160C94C741C051011112233445566778899D0AD1111D0AD03107461646F2D7468726561642D3464656101024DEA0410124810E2215110AFD6BB9215A6BFAC530C0402A0F7F8 63144777 2725 --paa-trust-store-path /home/ak/work/projects/Github/connectedhomeip/credentials/production/paa-root-certs
```

For NRF DK: The same as above except for the changed device discriminator, passcode

The commissioning times out every single time consistently for me. A co-worker is unable to reproduce the failure (only difference being his commissioner is running Gentoo, while I've tried and encountered the same failure across RPi, Ubuntu 22.04, Ubuntu 20.04)

Logs attached, would love some pointers to investigate further. Thanks in advance!
[chip-tool.log](https://github.com/project-chip/connectedhomeip/files/12373239/chip-tool.log)

### Bug prevalence

N/N instances

### GitHub hash of the SDK that was being used

e1f0ffd79b170a0ba0b8e6bf0a85767b59b5de9e

### Platform

nrf

gokul-bhalodiya-7, 2023-08-17:
As per the logs, thread border router failed at "FindOperational" commissioning stage.

So that it looks like there's a problem in the process of discovering and connecting to the operational dataset (the network configuration) of the Matter network.

You should check that the dataset value given as argument to the chip-tool **hex:0E080000000000010000000300000F35060004001FFFE00208D0AD1111D0AD22220708FD993160C94C741C051011112233445566778899D0AD1111D0AD03107461646F2D7468726561642D3464656101024DEA0410124810E2215110AFD6BB9215A6BFAC530C0402A0F7F8** is matching with the dataset value at the thread border router or not.

You can display openthread dataset value by running command "ot-ctl dataset active -x" at thread border router side and give this value to the chip-tool as dataset value argument.

Having same network configuration settings should resolve the issue.

adithyakirank, 2023-08-17:
I reprogrammed the ESP thread border router and confirmed that I am using the right dataset. The logs with the new + confirmed dataset are attached below, the commissioning still fails though.

[chip-tool-2.log](https://github.com/project-chip/connectedhomeip/files/12373718/chip-tool-2.log)

gokul-bhalodiya-7, 2023-08-17:
Can you please confirm the output of the _**ot-ctl state**_ and _**ot-ctl dataset active -x**_ commands

adithyakirank, 2023-08-18:
I confirmed the state and the dataset. The state was `leader` and the active dataset matched the one I am using in my commands. However when running the commissioning, the border router logs started to look like this when the `FindOperational` stage started to fail/timeout repeatedly:

```
I (307007) OPENTHREAD: Host delete
I (307007) OPENTHREAD: delete host 2A1878F8326DFBDA
E (307007) OPENTHREAD: Failed to Send UDP message, err: -4
I (307407) OPENTHREAD: Received PIO
I (307407) OPENTHREAD: Received RIO
I (307407) OPENTHREAD: prefix :: lifetime 180

I (309187) OPENTHREAD: Host delete
I (309187) OPENTHREAD: delete host 2A1878F8326DFBDA
E (309187) OPENTHREAD: Failed to Send UDP message, err: -4
I (310477) OPENTHREAD: Received PIO
I (310477) OPENTHREAD: Received RIO
I (310477) OPENTHREAD: prefix :: lifetime 180

I (312637) OPENTHREAD: Host delete
I (312637) OPENTHREAD: delete host 2A1878F8326DFBDA
E (312637) OPENTHREAD: Failed to Send UDP message, err: -4
I (313447) OPENTHREAD: Received PIO
I (313447) OPENTHREAD: Received RIO
I (313447) OPENTHREAD: prefix :: lifetime 180

I (316417) OPENTHREAD: Received PIO
I (316417) OPENTHREAD: Received RIO
I (316417) OPENTHREAD: prefix :: lifetime 180

I (318547) OPENTHREAD: Host delete
I (318547) OPENTHREAD: delete host 2A1878F8326DFBDA
E (318557) OPENTHREAD: Failed to Send UDP message, err: -4
I (319487) OPENTHREAD: Received PIO
I (319487) OPENTHREAD: Received RIO
I (319487) OPENTHREAD: prefix :: lifetime 180

I (322457) OPENTHREAD: Received PIO
I (322457) OPENTHREAD: Received RIO
I (322457) OPENTHREAD: prefix :: lifetime 180

I (327837) OPENTHREAD: Host delete
I (327837) OPENTHREAD: delete host 2A1878F8326DFBDA
E (327837) OPENTHREAD: Failed to Send UDP message, err: -4
```

adithyakirank, 2023-08-18:
With a more detailed level the debug logs on the border router look like this:

```
I(74912) OPENTHREAD:[I] SrpServer-----: Processed SRP update info
I(74912) OPENTHREAD:[I] SrpServer-----:     Host:2A1878F8326DFBDA.default.service.arpa.
I(74932) OPENTHREAD:[I] SrpServer-----:     Lease:0, key-lease:1209600, ttl:7200
I(74932) OPENTHREAD:[I] SrpServer-----:     No host address
I(74942) OPENTHREAD:[I] SrpServer-----: SRP update handler is notified (updatedId = 703803703)
I (74942) OPENTHREAD: Host delete
I (74952) OPENTHREAD: delete host 2A1878F8326DFBDA
I(74952) OPENTHREAD:[I] SrpServer-----: Handler result of SRP update (id = 703803703) is received: OK
I(74972) OPENTHREAD:[I] SrpServer-----: Committing update for existing host 2A1878F8326DFBDA.default.service.arpa.
I(74982) OPENTHREAD:[I] SrpServer-----:     Granted lease:0, key-lease:680400, ttl:30
E (74992) OPENTHREAD: Failed to Send UDP message, err: -4
I(74992) OPENTHREAD:[I] SrpServer-----: Send success response with granted lease: 0 and key lease: 680400
I(75012) OPENTHREAD:[I] RoutingManager: Received RA from fe80:0:0:0:10:18ff:fea7:6544 on infra netif 4
I(75012) OPENTHREAD:[I] RoutingManager: - RA Header - default route - lifetime:180
I(75022) OPENTHREAD:[I] RoutingManager: - PIO 2a02:810d:400:23f4::/64 (valid:300, preferred:300)
I(75032) OPENTHREAD:[I] RoutingManager: - RIO ::/0 (lifetime:180, prf:medium)
I(75042) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(75052) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(75232) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(75422) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(75602) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(75792) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(75982) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(76172) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(76362) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(76542) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(76742) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(76942) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(77142) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(77342) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(77542) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(77752) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(77952) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(78142) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(78352) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(78542) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(78752) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(78952) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(79142) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(79352) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-25, ack-fp:0
I(79552) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(79752) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(79952) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(80152) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(80352) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(80552) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(80752) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(80782) OPENTHREAD:[I] MeshForwarder-: Received IPv6 UDP msg, len:133, chksum:1b31, ecn:no, from:0x5401, sec:yes, prio:normal, rss:-26.0
I(80792) OPENTHREAD:[I] MeshForwarder-:     src:[fd22:4062:9a5a:1:ef26:f6b1:9410:2edf]:49155
I(80802) OPENTHREAD:[I] MeshForwarder-:     dst:[fd22:4062:9a5a:2:0:0:22fe:d46]:5684
I(80952) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(81152) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(81352) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(81552) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(81752) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(81952) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(82152) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
I(82352) OPENTHREAD:[I] DataPollHandlr: Rx data poll, src:0x5401, qed_msgs:0, rss:-26, ack-fp:0
[... cut ...]
I(90412) OPENTHREAD:[I] SrpServer-----: SRP update handler is notified (updatedId = 703803704)
I (90422) OPENTHREAD: Host delete
I (90422) OPENTHREAD: delete host 2A1878F8326DFBDA
I(90432) OPENTHREAD:[I] SrpServer-----: Handler result of SRP update (id = 703803704) is received: OK
I(90442) OPENTHREAD:[I] SrpServer-----: Committing update for existing host 2A1878F8326DFBDA.default.service.arpa.
I(90452) OPENTHREAD:[I] SrpServer-----:     Granted lease:0, key-lease:680400, ttl:30
E (90462) OPENTHREAD: Failed to Send UDP message, err: -4
```

adithyakirank, 2023-08-18:
Since chip tool logged this during the commissioning failure:
```
[1692348106.653561][305935:305937] CHIP:DIS: Checking node lookup status after 45002 ms
[1692348106.653624][305935:305937] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: operational discovery failed: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout
[1692348106.653636][305935:305937] CHIP:DIS: Retrying operational DNS-SD discovery. Attempts remaining: 2
[1692348106.654101][305935:305937] CHIP:CTL: Session establishment failed for <0000000000000001, 1>, error: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout.  Next retry expected to get a response to Sigma1 or fail within 60 seconds
```

I ran this `avahi-browse -a` to check what is in mDNS/DNS-SD (Not sure if this looks right for the border router):
```
+  wlan0 IPv6 ESP_IDF_Border_Router                             _meshcop._udp        local
+  wlan0 IPv4 ESP_IDF_Border_Router                             _meshcop._udp        local
+  wlan0 IPv6 Brother DCP-L2550DN series                    Internet Printer     local
+  wlan0 IPv4 Brother DCP-L2550DN series                    Internet Printer     local
```

bzbarsky-apple, 2023-08-28:
> I ran this avahi-browse -a to check what is in mDNS/DNS-SD

That's definitely not showing any Matter operational advertisements.....  So the discovery failure by chip-tool is expected given that.

suktry, 2024-01-09:
Hello, I encountered the same issue as you. Could you please share how you resolved it? Any guidance you can provide would be appreciated. Thank you!