project-chip/connectedhomeip #28747: [1.1]  ESP32 TC-RR-1.1  Step 2 err

cheng5230, 2023-08-18:
### Reproduction steps

[MatterTest] 08-18 03:18:15.589 ERROR Exception occurred in test_TC_RR_1_1.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 775, in exec_one_test
    test_method()
  File "/root/python_testing/matter_testing_support.py", line 642, in async_runner
    return asyncio.run(body(*args, **kwargs))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/root/python_testing/TC_RR_1_1.py", line 163, in test_TC_RR_1_1
    await client.SendCommand(self.dut_node_id, 0, Clusters.OperationalCredentials.Commands.UpdateFabricLabel(label))
  File "/usr/local/lib/python3.10/dist-packages/chip/ChipDeviceCtrl.py", line 866, in SendCommand
    return await future
chip.interaction_model.InteractionModelError: InteractionModelError: Failure (0x1)
[MatterTest] 08-18 03:18:15.599 INFO [Test] test_TC_RR_1_1 ERROR

### Bug prevalence

app 

### GitHub hash of the SDK that was being used

13f5e8b012d76b06eb4918b3a618615253fc151f

### Platform

esp32

### Platform Version(s)

1.1

### Type

Test Improvement

VaishaliAvhale, 2023-08-30:
Please provide detailed logs and steps you followed to execute this test case.

dhrishi, 2024-10-25:
Closing due to lack of updates

closed 2024-10-25 by dhrishi