project-chip/connectedhomeip #28750: [Feature] Represent 64-bit Unsigned Integers as Strings in JSON Responses

TheWalkingDead1024, 2023-08-18:
### Feature description

Hello,

I would like to propose a suggestion for handling 64-bit unsigned integers in the JSON responses of this project.

As you may know, some programming languages, such as Java and Dart, do not natively support 64-bit unsigned integers. When these languages parse a JSON response containing a 64-bit unsigned integer that exceeds the maximum value of a 64-bit signed integer, they typically represent the value as a double. However, this can lead to a loss of precision, which is undesirable for many applications.

To address this issue, I suggest representing 64-bit unsigned integers as strings in JSON responses. This would allow developers using languages like Java and Dart to easily parse these values as BigInteger (or equivalent) objects, without losing any precision.

For example, instead of:

json
Copy code
{
  "node_id": 18446744073709551615
}
The JSON response could be:

json
Copy code
{
  "node_id": "18446744073709551615"
}
This way, the Matter Chip's underlying system can continue to work with these values as strings, and developers can choose to convert these strings to the appropriate numerical type in their own applications as necessary.

I believe this change would make the Matter Chip project more accessible and user-friendly for developers working in a variety of programming languages.

Thank you for considering this suggestion. I look forward to hearing your thoughts on this matter.

### Platform

all, android

bzbarsky-apple, 2023-08-28:
@TheWalkingDead1024 how is this different from https://github.com/project-chip/connectedhomeip/issues/25934?

Are there specific JSON strings being produced by in-tree Matter code that have this problem?

TheWalkingDead1024, 2023-08-31:
> 

sorry，I apologize for the inconvenience. I've realized that this issue is a duplicate of https://github.com/project-chip/connectedhomeip/issues/25934. It was unintentional, and I'm sorry for any confusion I may have caused.

bzbarsky-apple, 2023-09-02:
@TheWalkingDead1024 no need to apologize.  Duplicates happen; I've filed a fair number in my life.

bzbarsky-apple, 2023-09-02:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/25934

closed 2023-09-02 by bzbarsky-apple