project-chip/connectedhomeip #28776: [1.1] [K32W] [NXP] [lighting-app] lighting-app example compilation issue

gokul-2000, 2023-08-21:
### Reproduction steps

**CASE 1:**

I followed the guide given at link: [https://github.com/project-chip/connectedhomeip/tree/v1.1-branch/examples/lighting-app/nxp/k32w/k32w0](url)

I used NXP K32W USB dongle as matter end-device to run lighting-app example.

As per the guide, I used NXP SDK v2.6.11 and lighting-app example from connectedhomeip.

In that, I faced compilation errors for the unused variables and function parameters not matching, which I resolved but still, I am facing below mentioned error.

```
ninja -C out/debug
ninja: Entering directory `out/debug'
[1/8] ld ./chip-k32w0x-light-example
FAILED: chip-k32w0x-light-example chip-k32w0x-light-example.map 
arm-none-eabi-g++ -T../../third_party/connectedhomeip/examples/platform/nxp/k32w/k32w0/app/ldscripts/chip-k32w0x-linker.ld -Wl,-print-memory-usage -Wl,--defsym -Wl,__app_load_address__=0x4000 -Wl,--defsym -Wl,__app_stated_size__=0x99200 -march=armv7e-m -mcpu=cortex-m4 -mfloat-abi=soft -mthumb -Og --specs=nosys.specs --specs=nano.specs -Werror -Wl,--fatal-warnings -fdiagnostics-color -Wl,--gc-sections -T ../../third_party/connectedhomeip/third_party/pigweed/repo/pw_tokenizer/pw_tokenizer_linker_sections.ld -Wl,--wrap=malloc -Wl,--wrap=free -Wl,--wrap=realloc -Wl,--wrap=calloc -Wl,--wrap=MemoryAlloc -Wl,--wrap=_malloc_r -Wl,--wrap=_realloc_r -Wl,--wrap=_free_r -Wl,--wrap=_calloc_r -L../../third_party/connectedhomeip/third_party/pigweed/repo/pw_tokenizer -Wl,-Map,./chip-k32w0x-light-example.map @./chip-k32w0x-light-example.rsp -o ./chip-k32w0x-light-example
/home/gokul/BAE/matter-migration/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: ./chip-k32w0x-light-example section .text' will not fit in region Flash640'
/home/gokul/BAE/matter-migration/matter/.environment/cipd/packages/arm/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: region `Flash640' overflowed by 97976 bytes
Memory region         Used Size  Region Size  %age Used
        Flash640:      725176 B     627200 B    115.62%
     SCRATCH_RAM:          1 KB         1 KB    100.00%
            RAM0:       83536 B        87 KB     93.77%
            RAM1:         64 KB        64 KB    100.00%
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

-------------------------------------------------------------------------------------------------------

**CASE 2:**

I checkout the master branch commit id: 40fe1c6ac841a1520ac7c3885451674562491e36

I used NXP K32W USB dongle as matter end-device to run lighting-app example.

Followed link: https://github.com/project-chip/connectedhomeip/tree/40fe1c6ac841a1520ac7c3885451674562491e36

As per the guide, I used NXP SDK v2.6.11 and lighting-app example from connectedhomeip, I successfully compiled the example application.

After that, I flashed the bootloader binary(k32w061dk6_ssbl.bin) and lighting-app binary(chip-k32w0x-light-example.bin), but it's not displaying logs on the terminal which can verify that it's working or not.

-------------------------------------------------------------------------------------------------------

Let me know in case of more information is required.

Thanks in advance for any support. 

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

v1.1-branch

### Platform

k32w

### Type

Platform Issue

gokul-2000, 2023-08-22:
Hi @bzbarsky-apple,

Any help or guide would be much appreciable.

bzbarsky-apple, 2023-08-28:
For your case 1, you're running out of space on the device, right?  From the log you pasted:

> Region `Flash640' overflowed by 97976 bytes

Past that, I've never tried actually compiling for k32w.  You might want to ask @doru91 .

bzbarsky-apple, 2023-09-02:
@gokul-2000 I have no idea, sorry...

mihai-ignat, 2024-04-25:
Hello, @gokul-2000 .

Regarding the initial post, is case 1 still an issue?
For case 2, the binary generated by the build is not suitable for K32W0 USB dongle. It was made for the [K32W061DK6](https://www.nxp.com/products/wireless-connectivity/multiprotocol-mcus/advanced-development-kit-for-k32w061-and-jn5189-88:IOTZTB-DK006) board. Can you please try using K32W061DK6?

And for the last comment, regarding hardware related changes, the file defining the used pins and their function is pin_mux.c, which can be found in the K32W061 SDK, at path boards/k32w061dk6/wireless_examples/openthread/reed/bm/pin_mux.c.
This is where you need to start in order to adapt to your hardware configuration.

For faster responses, I suggest using the [NXP Wireless Connectivity community](https://community.nxp.com/t5/Wireless-Connectivity/bd-p/wireless-connectivity).

BR,
Mihai