project-chip/connectedhomeip #28796: [1.1] In interactive server mode, subscribing to too many attributes at once will cause a timeout

changzhe-github, 2023-08-22:
### Reproduction steps

1. Connect the Matter device to the Matter network of the Matter gateway
2. Use command on Matter gateway : chip-tool interactive server --port 9003 --iface null
3. Connect websocket and send commad: any subscribe-by-id 0xFFFFFFFF 0xFFFFFFFF 0 10 1001 3

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

910be1fcff2211e7ea65bee4fa62fea564909ea5

### Platform

raspi

### Type

Common Cluster Logic

changzhe-github, 2023-08-23:
[1692671855.030156][19508:19525] CHIP:DMG: Subscription established with SubscriptionID = 0x82a63660 MinInterval = 0s MaxInterval = 10s Peer = 03:00000000000003E9
[1692671855.030219][19508:19525] CHIP:DMG: MoveToState ReadClient[0x56a93ee0]: Moving to [Subscripti]
[1692671855.034258][19508:19525] CHIP:DMG: Refresh LivenessCheckTime for 14224 milliseconds with SubscriptionId = 0x82a63660 Peer = 03:00000000000003E9
[1692671883.836392][19508:19525] CHIP:EM: <<< [E:33391i S:10527 M:73228097 (Ack:66671197)] (S) Msg TX to 3:00000000000003E9 [5F10] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1692671883.836532][19508:19525] CHIP:IN: (S) Sending msg 73228097 on secure session with LSID: 10527
[1692671883.836759][19508:19525] CHIP:EM: Flushed pending ack for MessageCounter:66671197 on exchange 33391i

[1692671883.837077][19508:19525] CHIP:IN: Received a duplicate message with MessageCounter:66671197 on exchange 33391i
[1692671883.837169][19508:19525] CHIP:EM: >>> [E:33391i S:10527 M:66671197 (Ack:73228096)] (S) Msg RX from 3:00000000000003E9 [5F10] --- Type 0001:04 (IM:SubscribeResponse)
[1692671883.837259][19508:19525] CHIP:EM: Generating StandaloneAck via exchange: 33391i
[1692671883.837311][19508:19525] CHIP:EM: Forcing tx of solitary ack for duplicate MessageCounter:66671197 on exchange 33391i
[1692671883.837494][19508:19525] CHIP:EM: <<< [E:33391i S:10527 M:73228098 (Ack:66671197)] (S) Msg TX to 3:00000000000003E9 [5F10] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1692671883.837600][19508:19525] CHIP:IN: (S) Sending msg 73228098 on secure session with LSID: 10527
[1692671883.838084][19508:19525] CHIP:IN: Received a duplicate message with MessageCounter:66671197 on exchange 33391i
[1692671883.838180][19508:19525] CHIP:EM: >>> [E:33391i S:10527 M:66671197 (Ack:73228096)] (S) Msg RX from 3:00000000000003E9 [5F10] --- Type 0001:04 (IM:SubscribeResponse)
[1692671883.838256][19508:19525] CHIP:EM: Generating StandaloneAck via exchange: 33391i
[1692671883.838959][19508:19525] CHIP:EM: Forcing tx of solitary ack for duplicate MessageCounter:66671197 on exchange 33391i
[1692671883.839406][19508:19525] CHIP:EM: <<< [E:33391i S:10527 M:73228099 (Ack:66671197)] (S) Msg TX to 3:00000000000003E9 [5F10] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1692671883.839924][19508:19525] CHIP:IN: (S) Sending msg 73228099 on secure session with LSID: 10527
[1692671883.840625][19508:19525] CHIP:IN: Received a duplicate message with MessageCounter:66671197 on exchange 33391i
[1692671883.841546][19508:19525] CHIP:EM: >>> [E:33391i S:10527 M:66671197 (Ack:73228096)] (S) Msg RX from 3:00000000000003E9 [5F10] --- Type 0001:04 (IM:SubscribeResponse)
[1692671883.841955][19508:19525] CHIP:EM: Generating StandaloneAck via exchange: 33391i
[1692671883.842321][19508:19525] CHIP:EM: Forcing tx of solitary ack for duplicate MessageCounter:66671197 on exchange 33391i
[1692671883.842728][19508:19525] CHIP:EM: <<< [E:33391i S:10527 M:73228100 (Ack:66671197)] (S) Msg TX to 3:00000000000003E9 [5F10] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1692671883.843211][19508:19525] CHIP:IN: (S) Sending msg 73228100 on secure session with LSID: 10527
[1692671883.844275][19508:19525] CHIP:IN: Received a duplicate message with MessageCounter:66671197 on exchange 33391i
[1692671883.844864][19508:19525] CHIP:EM: >>> [E:33391i S:10527 M:66671197 (Ack:73228096)] (S) Msg RX from 3:00000000000003E9 [5F10] --- Type 0001:04 (IM:SubscribeResponse)
[1692671883.847825][19508:19525] CHIP:EM: Generating StandaloneAck via exchange: 33391i
[1692671883.847871][19508:19508] CHIP:TOO: LWS_CALLBACK_SERVER_WRITEABLE
[1692671883.848206][19508:19525] CHIP:EM: Forcing tx of solitary ack for duplicate MessageCounter:66671197 on exchange 33391i
[1692671883.849243][19508:19525] CHIP:EM: <<< [E:33391i S:10527 M:73228101 (Ack:66671197)] (S) Msg TX to 3:00000000000003E9 [5F10] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1692671883.849831][19508:19525] CHIP:IN: (S) Sending msg 73228101 on secure session with LSID: 10527
[1692671884.814399][19508:19508] CHIP:TOO: LWS_CALLBACK_SERVER_WRITEABLE

[1692671898.073875][19508:19525] CHIP:DMG: Subscription Liveness timeout with SubscriptionID = 0x82a63660, Peer = 03:00000000000003E9
[1692671898.073982][19508:19525] CHIP:IN: SecureSession[0x56a43f10]: MarkAsDefunct Type:2 LSID:10527
[1692671898.074028][19508:19525] CHIP:SC: SecureSession[0x56a43f10]: Moving from state 'kActive' --> 'kDefunct'
[1692671898.074066][19508:19525] CHIP:DMG: MoveToState ReadClient[0x56a93ee0]: Moving to [      Idle]
[1692671898.074115][19508:19525] CHIP:TOO: Error: ../chip/examples/chip-tool/third_party/connectedhomeip/src/app/ReadClient.cpp:894: CHIP Error 0x00000032: Timeout
[1692671898.074174][19508:19525] CHIP:TOO: Run command failure: ../chip/examples/chip-tool/third_party/connectedhomeip/src/app/ReadClient.cpp:894: CHIP Error 0x00000032: Timeout

bzbarsky-apple, 2023-08-28:
@changzhe-github Hard to say why you are never getting a report without seeing the server log, but something is definitely up with your connectivity.  Observe all the retransmits of the SubscribeResponse message, which suggests the other side is not getting the acks....

I strongly suggest attaching (not pasting) logs from both sides of the interaction so we can try to figure out what the server is seeing/doing here.