project-chip/connectedhomeip #28797: [CERT-TEST-FAILURE] [TC-DGGEN-2.1] - Test Failed

Rajashreekalmane, 2023-08-22:
### Feature Area

Other

### Test Case

TC-DGGEN-2.1

### Reproduction steps

**As per Spec and Test-plan Expectation after Factory Reset the DUT, Reboot Count attribute value should be set to 0 instead of 1** 

Execute the below mentioned command to put DUT into a commissionable state,
  step-1:  create a file using touch command , something like touch mytest.txt
  step-2:  chmod 777 mytest.txt
  step-3:  launch the app`  ./chip-all-clusters-app --KVS ./mytest.txt
`
Once DUT reach the  commissionable state, sent below mentioned command on TH.
` ./chip-tool pairing onnetwork 1 20202021`
After a successfully paired sent mentioned command from TH

**Verifying the Step 3(reboot-count attribute)** 

1 .TH reads a RebootCount attribute value from DUT.
 `./chip-tool generaldiagnostics read reboot-count 1 0`

On TH(chip-tool), Verify that RebootCount attribute data type must be UINT16
>[1692697157.169409][4935:4937] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0033 Attribute 0x0000_0001 DataVersion: 629105351
**[1692697157.169523][4935:4937] CHIP:TOO:   RebootCount: 1**

2. after reading  RebootCount attribute reboot the DUT, by killing DUT more than 2 times and launch again, without commission follow the next steps

./chip-tool generaldiagnostics read reboot-count 1 0
The  reboot-count count should match with the number of rebooted times
> [1692697170.581354][4938:4940] CHIP:DMG: }
> [1692697170.581583][4938:4940] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0033 Attribute 0x0000_0001 DataVersion: 2251393556
> [1692697170.581692][4938:4940] CHIP:TOO:   RebootCount: 4

3.  Factory reset on DUT using the mentioned command and launch again, without commission follow the next steps
 `sudo rm -rf /tmp/chip_*`

`./chip-tool generaldiagnostics read reboot-count 1 0``
On TH(chip-tool), Verify that RebootCount attribute  value set to 0

> [1692697194.341566][4942:4944] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0033 Attribute 0x0000_0001 DataVersion: 3812177057
> [1692697194.341654][4942:4944] CHIP:TOO:   RebootCount: 1

As per Spec and Test-plan Expectation after Factory Reset the DUT, Reboot Count attribute value -
**Expected value**  : 0
**Actual value** : 1

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

5ca8bec0a813c51f82a55da4b8bfffa32546248e

### Platform

raspi

### Anything else?

As per Spec : 

> The RebootCount attribute SHALL indicate a best-effort count of the number of times the Node has rebooted. The RebootCount attribute SHOULD be incremented each time the Node reboots. The RebootCount attribute SHALL NOT be incremented when a Node wakes from a low-power or sleep state. The RebootCount attribute SHALL only be reset to 0 upon a factory reset of the Node.

and Test-plan expecting  : 

> The data type must be UINT16 and the count should match with the number of rebooted times. A low power or sleep shall not count as a reboot. **Upon the factory reset, the count shall be set to 0**.

As per Spec and Test-plan Expectation after Factory Reset the DUT, Reboot Count attribute value should be set to 0 instead of 1

**Execution Logs :** 
[TC-DGGEN-2.1_DUT.txt](https://github.com/project-chip/connectedhomeip/files/12407829/TC-DGGEN-2.1_DUT.txt)
[TC-DGGEN-2.1_TH.txt](https://github.com/project-chip/connectedhomeip/files/12407831/TC-DGGEN-2.1_TH.txt)

**Test-plan Reference :** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/general_diagnostics.adoc#tc-dggen-2-1-attributes-dut_server

**Spec Reference :** 
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/DiagnosticsGeneral.adoc#rebootcount-attribute

tcarmelveilleux, 2023-08-24:
After factory reset, it should be either 1 or 0, doesn't matter. As long as subsequent reboots increment, we are good. I think "reboot count" can be interpreted differently by different people since it's unclear in spec. So the reboot count is set to 0 after factory reset, but can it be set to 1 during first boot? That's the question...

tcarmelveilleux, 2023-08-24:
Can you please check Linux platform implementation, in the configuration manager, what the impl is?

andy31415, 2023-08-24:
For me it seems to make sense that "I reboot, so I increase the counter" results in "first time reboot (after factory reset), the counter is 1 ... since i rebooted". I don't think trying to make sure it is 0 after factory reset is worth the extra effort. If spec demands that we should fix the spec.

andy31415, 2023-08-24:
"The RebootCount attribute SHALL only be reset to 0 upon a factory reset of the Node." I interpret this as factory reset sets it to 0. However if implementation reboots after factory reset (I imagine most do) then we reboot and the count then should be 1.

We should fix our tests.

closed 2023-08-25 by cjandhyala