project-chip/connectedhomeip #28821: [BUG] removing events from zap file doesn't take effect

step0035, 2023-08-23:
### Reproduction steps

There's a known issue with the zap gui whereby after enabling events in the gui, the next time you launch the gui, the event will still show up as disabled.
So I manually modified the zap file to exclude event 0x4 of TimeSync by setting `include` to 0.
However, the event still shows up in the generated `endpoint_config.h` and running the timesync-1-1 test case will still read that the event 0x4 is enabled.
Tested on all-clusters-app linux.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

raspi

### Anything else?

[timesync-1-1_2023_08_23_11_30_00.log](https://github.com/project-chip/connectedhomeip/files/12414892/timesync-1-1_2023_08_23_11_30_00.log)

bzbarsky-apple, 2023-08-28:
@step0035 Please report this in https://github.com/project-chip/zap/issues

step0035, 2023-08-29:
https://github.com/project-chip/zap/issues/1114

closed 2023-08-29 by step0035