project-chip/connectedhomeip #28847: [Build] Fuschia installation error in Mac OS - running app in docker container

ymrudula, 2023-08-23:
### Build issue(s)

1.Open connectedhomeip application in a docker container in VS code
2. Run the activate.sh shell script
3. Get error failed to install Fuschia

Host OS: mac OS with intel chip. running big sur. docker container running on linux. 

see error log below -

```
[P3522 17:06:42.453 deployer.go:409 I] [unzip  7/12] Deployed fuchsia/third_party/clang/linux-amd64:AD59JKqO1TWgxp7uA0Fq99aIrVCXG2i4HShWi6sFFV8C in 96.0s
Errors:
  failed to install fuchsia/third_party/armgcc/linux-amd64 in "arm": adding files to the site root: lstat /workspaces/connectedhomeip/.environment/cipd/packages/.cipd/pkgs/0/9z1FnQUHn-LGwcPNIsxnZzTE3HjoopYzwWlK3f1-HG0C/share/doc/gdb/qMemTags.html: no such file or directory
/workspaces/connectedhomeip/.environment/cipd/cipd ensure -ensure-file /workspaces/connectedhomeip/.environment/cipd/packages.ensure -root /workspaces/connectedhomeip/.environment/cipd/packages -log-level debug -json-output /workspaces/connectedhomeip/.environment/cipd/packages.json -cache-dir /home/vscode/.cipd-cache-dir -max-threads 0
[\]
None
Traceback (most recent call last):
  File "/workspaces/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 959, in <module>
    sys.exit(main())
  File "/workspaces/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 951, in main
    return EnvSetup(**vars(parse())).setup()
  File "/workspaces/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 558, in setup
    result = step(spin)
  File "/workspaces/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 671, in cipd
    if not cipd_update.update(
  File "/workspaces/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 382, in update
    subprocess.check_call(cmd, stdout=outs, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/workspaces/connectedhomeip/.environment/cipd/cipd', 'ensure', '-ensure-file', '/workspaces/connectedhomeip/.environment/cipd/packages.ensure', '-root', '/workspaces/connectedhomeip/.environment/cipd/packages', '-log-level', 'debug', '-json-output', '/workspaces/connectedhomeip/.environment/cipd/packages.json', '-cache-dir', '/home/vscode/.cipd-cache-dir', '-max-threads', '0']' returned non-zero exit status 1.
Installing pip requirements for all...
DEPRECATION: distro-info 0.23ubuntu1 has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of distro-info or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
```

...

### Platform

ameba, vscode

Goks1730, 2023-10-26:
same issue 
do find the solution ?

ymrudula, 2023-10-27:
unfortunately no. 

I tried using a different commit and it seems to work one time randomly.

Goks1730, 2023-10-27:
for me the build was successful in Virtual Box Ubuntu
any idea how to flash any one example to board?

RobinKastberg, 2023-12-08:
I encountered this issue on /mnt/d on WSL but not in ~. Maybe long file paths or somesuch?

woody-apple, 2024-04-03:
Closing older issues, trying to focus on current build failures

closed 2024-04-03 by woody-apple

chenliuyang1989, 2024-08-27:
The issue still exist @woody-apple