project-chip/connectedhomeip #28858: [CERT-TEST-FAILURE] [TC-IDM-7.1]

cheng5230, 2023-08-24:
### Feature Area

Area:AirPurifiers

### Test Case

TC-IDM-7.1

### Reproduction steps

Use the below commands to provision the DUT with 5 reference devices (Reference device is unique instance of a Chip-tool)
RD1, RD2, RD3, RD4, RD5 should be on separate, distinct fabrics. in RPI platform

execute the below mentioned command to put DUT into a commissionable state, Pls use equivalent command on the respective DUT
./chip-all-clusters-app

Once DUT reach the  commissionable state pls send below mentioned command on RD's respectively. Pls use equivalent command on the respective DUT

### Bug prevalence

app 

### GitHub hash of the SDK that was being used

13f5e8b012d76b06eb4918b3a618615253fc151f

### Platform

esp32

### Anything else?

[1692753781.755873][2597:2597] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/Commands.cpp:78: CHIP Error 0x0000002F: Invalid argument
>>> 
     
>>> onoff subscribe on-time 100 1000 2 1 --commissioner-name beta --keepSubscriptions 1 

[1692753786.767387][2597:2602] CHIP:TOO: Sending command to node 0x2

[1692753786.767487][2597:2602] CHIP:CSM: FindOrEstablishSession: PeerId = [2:0000000000000002]

[1692753786.767548][2597:2602] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found

[1692753786.767625][2597:2602] CHIP:CTL: Found an existing secure session to [2:0000000000000002]!

[1692753786.767688][2597:2602] CHIP:CTL: OperationalSessionSetup[2:0000000000000002]: State change 1 --> 5

[1692753786.767787][2597:2602] CHIP:TOO: cluster 0x0000_0006, attribute: 0x0000_4001, endpoint 1

[1692753786.767860][2597:2602] CHIP:TOO: Sending SubscribeAttribute to:

[1692753786.768187][2597:2602] CHIP:IN: Prepared secure message 0xffff6c004f28 to 0x0000000000000002 (2)  of type 0x3 and protocolId (0, 1) on exchange 22973i with MessageCounter:146352729.

[1692753786.768286][2597:2602] CHIP:IN: Sending encrypted msg 0xffff6c004f28 with MessageCounter:146352729 to 0x0000000000000002 (2) at monotonic time: 00000000001EABCB msec

[1692753786.768783][2597:2602] CHIP:DMG: MoveToState ReadClient[0xffff6c010730]: Moving to [AwaitingIn]

[1692753787.062225][2597:2602] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:188350340 on exchange 22973i

[1692753787.062326][2597:2602] CHIP:EM: Found matching exchange: 22973i, Delegate: 0xffff6c010740

[1692753787.062409][2597:2602] CHIP:EM: Rxd Ack; Removing MessageCounter:146352729 from Retrans Table on exchange 22973i

[1692753787.062467][2597:2602] CHIP:EM: Removed CHIP MessageCounter:146352729 from RetransTable on exchange 22973i

[1692753808.790419][2597:2602] CHIP:IN: SecureSession[0xffff6c00fd30]: MarkAsDefunct Type:2 LSID:32712

[1692753808.790518][2597:2602] CHIP:SC: SecureSession[0xffff6c00fd30]: Moving from state 'kActive' --> 'kDefunct'

[1692753808.790584][2597:2602] CHIP:DMG: Time out! failed to receive report data from Exchange: 22973i

[1692753808.790639][2597:2602] CHIP:DMG: MoveToState ReadClient[0xffff6c010730]: Moving to [      Idle]

[1692753808.790714][2597:2602] CHIP:TOO: Error: ../../src/app/ReadClient.cpp:625: CHIP Error 0x00000032: Timeout

[1692753808.790785][2597:2602] CHIP:DMG: MoveToState ReadClient[0xffff6c010730]: Moving to [      Idle]

[1692753808.790988][2597:2597] CHIP:TOO: Run command failure: ../../src/app/ReadClient.cpp:625: CHIP Error 0x00000032: Timeout
>>> 

[TC-IDM-7.1_TH2.txt](https://github.com/project-chip/connectedhomeip/files/12425766/TC-IDM-7.1_TH2.txt)

cheng5230, 2023-08-24:
ESP32  DUT   CASE Session establishment failed: b 
[TC-IDM-7.1_DUT.txt](https://github.com/project-chip/connectedhomeip/files/12425779/TC-IDM-7.1_DUT.txt)

cjandhyala, 2023-08-24:
@dhrishi Could you pls have your team take a look at this issue ?

dhrishi, 2023-08-25:
Thanks for the tag @cjandhyala

@vaishaliavhale @PSONALl Please take a look

VaishaliAvhale, 2023-08-25:
Please use Interactive mode to Verify subscription test cases
Here is the command to enter into interactive mode:--   ./chip-tool interactive start

In each terminal, you need to commission a new fabric and send its corresponding command.
If you have any doubts please refer attached logs.
[IDM-7-1-TH.log](https://github.com/project-chip/connectedhomeip/files/12438120/IDM-7-1-TH.log)

Note: 
1. The Cluster and Commands should be based on the cluster implementation on the DUT.
2. You can use any supported chip cluster/attribute/command.

cheng5230, 2023-08-25:
This is the verification of TH-IDM-7.1 5 window logs
The platform used is **ESP32**. From the TH log error message, it appears that there is insufficient memory

[TC-IDM-7.1_TH5.txt](https://github.com/project-chip/connectedhomeip/files/12438667/TC-IDM-7.1_TH5.txt)
[TC-IDM-7.1_TH4.txt](https://github.com/project-chip/connectedhomeip/files/12438656/TC-IDM-7.1_TH4.txt)
[TC-IDM-7.1_TH1.txt](https://github.com/project-chip/connectedhomeip/files/12438651/TC-IDM-7.1_TH1.txt)
[TC-IDM-7.1_TH2.txt](https://github.com/project-chip/connectedhomeip/files/12438654/TC-IDM-7.1_TH2.txt)
[TC-IDM-7.1_TH3.txt](https://github.com/project-chip/connectedhomeip/files/12438655/TC-IDM-7.1_TH3.txt)

cheng5230, 2023-08-25:
> Please use Interactive mode to Verify subscription test cases Here is the command to enter into interactive mode:-- ./chip-tool interactive start
>
> In each terminal, you need to commission a new [...]

> Please use Interactive mode to Verify subscription test cases Here is the command to enter into interactive mode:-- ./chip-tool interactive start
>
> In each terminal, you need to commission a new [...]

This is the verification of TH-IDM-7.1 5 window logs
The platform used is **ESP32**. From the TH log error message,（CHIP_ERROR_NO_MEMORY） it appears that there is insufficient memory

[TC-IDM-7.1_TH5.txt](https://github.com/project-chip/connectedhomeip/files/12438667/TC-IDM-7.1_TH5.txt)
[TC-IDM-7.1_TH4.txt](https://github.com/project-chip/connectedhomeip/files/12438656/TC-IDM-7.1_TH4.txt)
[TC-IDM-7.1_TH1.txt](https://github.com/project-chip/connectedhomeip/files/12438651/TC-IDM-7.1_TH1.txt)
[TC-IDM-7.1_TH2.txt](https://github.com/project-chip/connectedhomeip/files/12438654/TC-IDM-7.1_TH2.txt)
[TC-IDM-7.1_TH3.txt](https://github.com/project-chip/connectedhomeip/files/12438655/TC-IDM-7.1_TH3.txt)

VaishaliAvhale, 2023-08-25:
Are you following any test case verification documents to test this test case?
If yes, please share them here.
As per my analysis of your log files, the precondition required to execute the test case isn't being met.

cheng5230, 2023-08-26:
> Are you following any test case verification documents to test this test case?
> If yes, please share them here.
> As per my analysis of your log files, the precondition required to execute the test case isn't being met.

**This is a test case**
[TC-IDM-7.1-CASE.xlsx](https://github.com/project-chip/connectedhomeip/files/12444329/TC-IDM-7.1-CASE.xlsx)

cheng5230, 2023-08-30:
> 

Version is 1.0

dhrishi, 2023-08-31:
@cjandhyala As the version is 1.0, can this be removed from cert blockers for v1.2 release?

VaishaliAvhale, 2023-08-31:
The latest verification document's version is v1.2 
@cheng5230 AirPurifier device type testing is not a part of release v1.0, It is in the v1.2 release. Therefore, please refer to the latest v1.2 test verification documents for the steps to test TC-IDM-7.1.