project-chip/connectedhomeip #28859: [CERT-TEST-FAILURE][TC-DGGEN-1.1]- Test Failed

Rajashreekalmane, 2023-08-24:
### Feature Area

Other

### Test Case

TC-DGGEN-1.1

### Reproduction steps

Execute the below mentioned command to put DUT into a commissionable state,
` ./chip-all-clusters-app --KVS ./mytest.txt`
Once DUT reach the commissionable state, sent below mentioned command on TH.
` ./chip-tool pairing onnetwork 1 20202021`
After a successfully paired sent mentioned command from TH

TH reads from the DUT the AcceptedCommandList attribute.
`./chip-tool generaldiagnostics read accepted-command-list 1 0 `
On TH(chip-tool), Verify that AcceptedCommandList attribute  list should contain 1 entry : 
**Expected value :**
> [1692861419.542251][6849:6851] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0033 Attribute 0x0000_FFF9 DataVersion: 2937068551
> [1692861419.542349][6849:6851] CHIP:TOO:   AcceptedCommandList: 1 entries
> [1692861419.542394][6849:6851] CHIP:TOO:     [1]: 0
> [1692861419.542694][6849:6851] CHIP:EM: <<< [E:25015i S:42251 M:37331830 (Ack:17135828)] (S) Msg TX to 1:0000000000000001 [853D] --- Type 0000:10 (SecureChannel:StandaloneAck)

Actual value :

> [1692861419.541960][6849:6851] CHIP:DMG: }
> [1692861419.542251][6849:6851] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0033 Attribute 0x0000_FFF9 DataVersion: 2937068551
> **[1692861419.542349][6849:6851] CHIP:TOO:   AcceptedCommandList: 2 entries
> [1692861419.542394][6849:6851] CHIP:TOO:     [1]: 0
> [1692861419.542429][6849:6851] CHIP:TOO:     [2]: 0**
> [1692861419.542694][6849:6851] CHIP:EM: <<< [E:25015i S:42251 M:37331830 (Ack:17135828)] (S) Msg TX to 

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

55327986b373b50a5aaa030deeaf05d5eef3317e

### Platform

raspi

### Anything else?

As per Spec and Test-Plan AcceptedCommandList attribute  list should contain only 1 entry.

**Spec Reference :** 
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/DiagnosticsGeneral.adoc#commands

**Test-plan Reference :** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/general_diagnostics.adoc#tc-dggen-1-1-global-attributes-with-dut_server

**PFA execution Log below :** 
[TC-DGGEN-1.1.txt](https://github.com/project-chip/connectedhomeip/files/12426682/TC-DGGEN-1.1.txt)

tcarmelveilleux, 2023-08-24:
Can you please re-run with `--trace-to json:log --trace_decode 1` appended to command line? There are insufficient logs to diagnose.

cecille, 2023-08-24:
Confirmed over here. This is handled in the ember-compatibility-functions.cpp global attributes reader. Looking into it

cecille, 2023-08-24:
This command appears to show up twice in the EmberAfCluster acceptedCommandList, implying this is likely a zap issue

cecille, 2023-08-24:
This is a zap issue.

From the endpoint_config.h for all clusters:
  /* Endpoint: 0, Cluster: General Diagnostics (server) */\
  /*   AcceptedCommandList (index=40) */ \
  0x00000000 /* TestEventTrigger */, \
  0x00000000 /* TestEventTrigger */, \
  chip::kInvalidCommandId /* end of list */, \

PR #28890 (2023-08-25): Manually remove client commant from server side
Somehow the server side of a cluster reported a client-side command, resulting in duplicate "accepted commands" and overall not being valid.

Unclear how this happens - I would expect UI to not allow such a thing. For now manual edit.

Fixes #28859
closed 2023-08-25 by andy31415