project-chip/connectedhomeip #28862: [CERT-TEST-FAILURE] - TC-WASHERCTRL-3.1 - Test failed

KishokG, 2023-08-24:
### Feature Area

Other

### Test Case

TC-WASHERCTRL-3.1

### Reproduction steps

1. Bring up the DUT with all-clusters-app.
2. Pair the DUT using chip-tool through "onnetwork" transport.
3. Run the command `./chip-tool laundrywashercontrols write number-of-rinses 5 1 1`

**Expected Outcome:**		
> [1692858657.915468][6723:6725] CHIP:DMG: 			StatusIB =
> [1692858657.915511][6723:6725] CHIP:DMG: 			{
> [1692858657.915570][6723:6725] CHIP:DMG: 				status = INVALID_IN_STATE,
> [1692858657.915639][6723:6725] CHIP:DMG: 			},

**Actual Outcome:**
> [1692858657.915468][6723:6725] CHIP:DMG: 			StatusIB =
> [1692858657.915511][6723:6725] CHIP:DMG: 			{
> [1692858657.915570][6723:6725] CHIP:DMG: 				status = 0x00 (SUCCESS),
> [1692858657.915639][6723:6725] CHIP:DMG: 			},

### Bug prevalence

Whenever I try to write NumberOfRinses with unsupported value

### GitHub hash of the SDK that was being used

55327986b373b50a5aaa030deeaf05d5eef3317e

### Platform

raspi

### Anything else?

-  In TC-WASHERCTRL-2.2, STEP 3 TH reads from the DUT the SupportedRinses attribute we get SupportedRinses: 2 entries [1, 2]. 
- so as per spec, write number-of-rinses should be able to toggle between 1 and 2 with SUCCESS response and If user attempt to set a value that is outside this supported range the DUT should respond with INVALID_IN_STATE. 
- But with current sample app we are able to write the value up to 255 without any issue.

crlonxp, 2023-08-25:
We use TOT source which include this  commit:
2e59655703 (HEAD -> master, origin/master, origin/HEAD) Add the Pre-Attribute checking for laundry washer controls cluster (#28618)
The test can pass. The log is attached

[issue-28862.zip](https://github.com/project-chip/connectedhomeip/files/12435835/issue-28862.zip)

KishokG, 2023-08-25:
Issue resolved in 2e59655703d901b1504b7d2e57f5d39316dc6a6d

closed 2023-08-25 by KishokG