project-chip/connectedhomeip #28925: [BUG] Error while building my dynamic bridge related project

mustafayuce33, 2023-08-28:
### Reproduction steps

I've used dynamic bridge app example as a base for my project and I had already get further and implemented a lot of device types. But now it seems that the example has been removed and looks like you stopped support for dynamic bridge as I've found this commit in the history; https://github.com/project-chip/connectedhomeip/pull/27077. What is the other way(s) you can show me to implement such a logic that I can add my devices dynamically in my code. 

And also here is the exception I get while I build my code;

[image]

I know this is because bridge device type was removed here https://github.com/project-chip/connectedhomeip/pull/27077/files#diff-6f94a762260f926e9253dd6382b03f9a439910cd004026103890d2babb1a6267L31. 

If I change the device type from bridge to another one, can I still continue using dynamic bridge as before? 

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commits/master

### Platform

raspi

bzbarsky-apple, 2023-08-28:
@andy31415

andy31415, 2023-08-29:
We removed dynamic bridge because there was no official maintainer for it and it started having issues needing resolving, mostly around new type support requirements for new clusters.

It can be added back if someone volunteers to maintain it.

mustafayuce33, 2023-08-30:
@andy31415 @bzbarsky-apple But I do not understand how non-matter devices will be bridged to matter fabric? The project/library, in its current form, addresses to devices with well defined endpoint(s) rather than bridged devices. I mean if your non-matter device(s) to be bridged to matter fabric have variable size then we do not have any chance to do that, if I understand correctly.

Then how do you think that other bridges such as philips hue, miHome will bridge their devices to matter fabric?

Or you just deleted example and you still support dynamic bridge logic in your code. What I mean is that may I continue using dynamic bridge logic If I move all those files generated by `code-gen` from `gen` folder to my `src` directory? Or you removed everything related with `dynamic bridge` stuff?

bzbarsky-apple, 2023-09-01:
@mustafayuce33 The sample is the thing that got removed.  All the dynamic endpoint logic is still there, bridge-app (which is different from dynamic-bridge-app in its approach) still exists and is tested as part of certification, multiple companies are in fact shipping bridges.

If you move all those files as you describe it might well work.

closed 2023-09-04 by mustafayuce33