project-chip/connectedhomeip #28927: [1.1] - Matter Controller is Unable to re-establish the sessions with End Device upon Reboot.

smishra-volansys, 2023-08-28:
### Reproduction steps

Take one Matter Controller and matter-supported device and follow the steps:

1. Pair the Matter Devices to the Controller.
2. Perform some actions on the device and monitor. (Plug on/off, Door open/close)
3. Power off the Matter Controller and Turn it ON after an Interval. 

Issue:
**When the Controller is powered ON again, it is unable to re-establish the Operational Sessions with the Paired Matter Devices.** 

We also made sure to check the Open Thread credentials as all the devices are supported. They are also included in the Open Thread Network. 
We were able to re-establish the controller-device connection in Matter v1.0-branch but facing an issue in v1.1.

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

v1.1-branch

### Platform

other

### Platform Version(s)

v1.1-branch

### Type

Platform Issue

### Anything else?

[Log_restart.txt](https://github.com/project-chip/connectedhomeip/files/12455727/Log_restart.txt)
PFA the Log file for reference.

bzbarsky-apple, 2023-08-28:
> it is unable to re-establish the Operational Sessions

Is the attached "Log_restart.txt" a controller log?  If so, that shows it trying to do operational instance lookups and failing:
```
[2023/08/28 14:42:36] [DIS] OperationalSessionSetup[1:0000000000000004]: operational discovery failed: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
...
[2023/08/28 14:42:36] [DIS] OperationalSessionSetup[1:0000000000000006]: operational discovery failed: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
...
etc
```

Does this controller, after restart, have connectivity to the relevant network?   Is it using platform dns-sd or minimal mdns?  What platform is it running on?

smishra-volansys, 2023-08-29:
Hi @bzbarsky-apple, Please find the answers below,

Does this controller, after restart, have connectivity to the relevant network? 
-> **Yes it does establish the connection with the Relevant network**

Is it using platform dns-sd or minimal mdns?
-> **It is using the Platform dns-sd**

What platform is it running on?
-> **We are running it on iMx6 ULL platform (Custom Linux).**

bzbarsky-apple, 2023-09-01:
So this is using avahi for DNS-SD?  Can you check whether avahi is seeing the device after reboot?

smishra-volansys, 2023-09-08:
We need to check that, but as far as our understanding, avahi should be able to see the device after reboot. 
In this case, we will check once and update.
JFYI, all our Matter Devices are ble-thread supported, hence they primarily rely on BLE for the commissioning.

smishra-volansys, 2023-09-11:
Hi @bzbarsky-apple, I can confirm that Avahi is able to see the device after Reboot. 
But when the device is trying to Re-establish the session we get the log as 

[2023/09/11 09:12:51] [DIS] SRV record already actively processed.
[2023/09/11 09:12:51] [DIS] SRV record already actively processed.
[2023/09/11 09:12:51] [DIS] SRV record already actively processed.

Does that mean, the Controller already has details about the Device from the previous state and it is trying to re-establish the new one but found it already existing in terms of SRV records?

We faced a similar issue during an active connection, the controller when rebooted didn't establish the session with the end device, which we resolved with some code changes and by making the configuration files accessible to the database. But we cannot access the end device code, hence this issue persists now when the end device is rebooted.

bzbarsky-apple, 2023-09-11:
That's a question for @andy31415 : that log indicates that minimal mdns is being used here.

andy31415, 2023-09-12:
Could you capture the DNSSD traffic to see if the queries are actually being answered?
The `SRV record already processed' just means that DNSSD is processing _some_ dnssd packet however it may not be the actual response to the device query or maybe the device answers with different data somehow.

I do not believe there is sufficient information in the logs to debug this - the logs seem to just say "looking for device but it does not reply to dnssd".

smishra-volansys, 2023-09-28:
Hi Team,
Thanks for the Responses. I am unable to trace the packets yet. 
Will Revert once done!

smishra-volansys, 2023-11-08:
Hi Team, 
Please Find attached the DNS Trace file for the Device, 
Here the Mac address ending with 6f34 is our Matter Controller. 
I am unable to upload the entire pcap file hence attaching a snap
[image]

bzbarsky-apple, 2023-11-13:
@andy31415 Please see above

andy31415, 2023-11-13:
@smishra-volansys could you give me instructions on how to reproduce with the SDK examples (e.g. chip-tool/repl and one of the thread devices ... I have efr32 and nordic devices at my desk).

smishra-volansys, 2023-11-16:
@andy31415 Sure! 

Steps To Reproduce this Issue:

1. Using Chip-tool or Chip Android App, pair an event-based Matter Device. 
2. Read a few events on the Controller end and make sure that the Commissioning has been done and the connection has been established. 
3. Now Power off the Matter Device and power it ON again. 
4. During this time, it will fail to establish the connection again and show the SRV records already actively processed.

Please make sure to subscribe to the events after commissioning so that you can see the events and the Logs

smishra-volansys, 2023-11-21:
@andy31415, were you able to reproduce using these steps?

smishra-volansys, 2023-12-06:
Hi @andy31415,
Is there any update on this?
@bzbarsky-apple

bzbarsky-apple, 2023-12-08:
I don't know anything about minimal mdns, sorry.