project-chip/connectedhomeip #28941: [CERT-TEST-FAILURE] DRLK-1.1 Cluster revision should be 7

sumaky, 2023-08-29:
### Feature Area

Other

### Test Case

DRLK-1.1

### Reproduction steps

 ./chip-tool doorlock read cluster-revision 1 1
ClusterRevision: 6

Cluster revision shows 6 but specification shows 7 Added support for European door locks (unbolt feature)

### Bug prevalence

Manual verification 

### GitHub hash of the SDK that was being used

NA

### Platform

other

### Anything else?

NA

Survensa, 2023-08-29:
Verified with commit : `cb83d1ef7376da8e0685e23f5285333c8ce3c4d4` but not able to reproduce the issue can you please cross verify and close the issue. 

```
[1693291931.023758][2656:2658] CHIP:DMG: ReportDataMessage =
[1693291931.023784][2656:2658] CHIP:DMG: {
[1693291931.023804][2656:2658] CHIP:DMG: 	AttributeReportIBs =
[1693291931.023837][2656:2658] CHIP:DMG: 	[
[1693291931.023860][2656:2658] CHIP:DMG: 		AttributeReportIB =
[1693291931.023891][2656:2658] CHIP:DMG: 		{
[1693291931.023916][2656:2658] CHIP:DMG: 			AttributeDataIB =
[1693291931.023945][2656:2658] CHIP:DMG: 			{
[1693291931.023974][2656:2658] CHIP:DMG: 				DataVersion = 0x46a604cd,
[1693291931.024003][2656:2658] CHIP:DMG: 				AttributePathIB =
[1693291931.024033][2656:2658] CHIP:DMG: 				{
[1693291931.024068][2656:2658] CHIP:DMG: 					Endpoint = 0x1,
[1693291931.024101][2656:2658] CHIP:DMG: 					Cluster = 0x101,
[1693291931.024132][2656:2658] CHIP:DMG: 					Attribute = 0x0000_FFFD,
[1693291931.024162][2656:2658] CHIP:DMG: 				}
[1693291931.024194][2656:2658] CHIP:DMG: 					
[1693291931.024225][2656:2658] CHIP:DMG: 				Data = 7, 
[1693291931.024253][2656:2658] CHIP:DMG: 			},
[1693291931.024283][2656:2658] CHIP:DMG: 			
[1693291931.024307][2656:2658] CHIP:DMG: 		},
[1693291931.024335][2656:2658] CHIP:DMG: 		
[1693291931.024357][2656:2658] CHIP:DMG: 	],
[1693291931.024386][2656:2658] CHIP:DMG: 	
[1693291931.024409][2656:2658] CHIP:DMG: 	SuppressResponse = true, 
[1693291931.024433][2656:2658] CHIP:DMG: 	InteractionModelRevision = 10
[1693291931.024455][2656:2658] CHIP:DMG: }
[1693291931.024586][2656:2658] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0101 Attribute 0x0000_FFFD DataVersion: 1185285325
[1693291931.024641][2656:2658] CHIP:TOO:   ClusterRevision: 7
```

sumaky, 2023-08-29:
ok looks like this is not an issue.

closed 2023-08-29 by sumaky