project-chip/connectedhomeip #28960: [Build] Android sample app

dzulhafiz-at, 2023-08-30:
### Build issue(s)

1. Compiling the connectedhomeip/examples/android code failed

All of the CHIP imports are not resolved. Sample imports are:
`import chip.devicecontroller.ChipDeviceController
import chip.devicecontroller.ClusterIDMapping.*
import chip.devicecontroller.InvokeCallback
import chip.devicecontroller.ReportCallback
import chip.devicecontroller.ResubscriptionAttemptCallback
import chip.devicecontroller.SubscriptionEstablishedCallback
import chip.devicecontroller.model.ChipAttributePath
import chip.devicecontroller.model.ChipEventPath
import chip.devicecontroller.model.InvokeElement
import chip.devicecontroller.model.NodeState
import chip.tlv.AnonymousTag
import chip.tlv.ContextSpecificTag
import chip.tlv.TlvWriter`

Extra information: I am using Windows 10 laptop

Steps I did:
Clone connectedhomeip
Open connectedhomeip/examples/android/CHIPTool in Android Studio
Install latest NDK (Side by Side) - 25.2.9519653
Install latest Command Line Tools
Install SDK 26
Add ANDROID_HOME and ANDROID_NDK_HOME into my Environment Variables
Add JAVA_HOME will following value "C:\Program Files\Android\Android Studio\jre"
Execute scripts/bootstrap.sh
Next I tried the "Building Android CHIPTool from scripts" step but the cmd provided there is not working, so I change to "python scripts\build\build_examples.py --target android-arm64-chip-tool build" but I am getting error:
`Traceback (most recent call last):
  File "C:\xxx\AndroidStudioProjects\connectedhomeip\scripts\build\build_examples.py", line 22, in <module>
    import click
ModuleNotFoundError: No module named 'click'`

But I assume this cmd would create an APK file, right? So, I return to Android Studio and try to run the examples/android/CHIPTool code, but still the CHIP import is not resolve and when I go ahead with "Run" the app, I got this:
`Caused by: java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.lang.String java.io.File.path accessible: module java.base does not "opens java.io" to unnamed module @d1886a8
`

Sorry if my question here shouldn't create an issue, but I had been stuck here for few days and still couldn't figure it out how to get this code working on my machine. I would like to Debug/Run this app so that I could put breakpoint etc to understand the flow better.

### Platform

android

### Anything else?

[image]
[image]

joonhaengHeo, 2023-08-30:
@dzulhafiz-at 
I didn't try android ChipTool build in windows environment.

If you load the SDK's example App (example/android/ChipTool) in android studio without library copy, I think this error appears because the library is not built.

dzulhafiz-at, 2023-08-30:
From this connectedhomeip source code, is there a way for me to import the code into Android Studio so I could play around with the Matter commissioning and sending cluster command etc from my Android phone?

vecoder, 2023-08-31:
@joonhaengHeo  

> I didn't try android ChipTool build in windows environment.

Do you recomand to use Linux system to build the Android sample app?

dzulhafiz-at, 2023-09-01:
> You can find android build guide below URL. https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md

Actually this is the guide that I am referring too. I did mentioned in "Steps I did:" above what I had done but having issue at  "Building Android CHIPTool from scripts" step. A bit busy now, will try again next week in Mac if able to proceed

yunhanw-google, 2023-09-01:
You can try the below,
docker run -it -v ~/workspace/connectedhomeip:/connectedhomeip ghcr.io/project-chip/chip-build-android:1

git config --global --add[ safe.directory](http://safe.directory/) /connectedhomeip
git config --global --add[ safe.directory](http://safe.directory/) /connectedhomeip/third_party/pigweed/repo

source scripts/bootstrap.sh
./scripts/build/build_examples.py --target linux-x64-java-matter-controller build
./scripts/build/build_examples.py --target linux-x64-all-clusters build
./scripts/build/build_examples.py --target linux-x64-chip-tool build

Qianweiyin, 2023-11-06:
@dzulhafiz-at Hello brother, has this problem been solved？

Mukhammadjon-Jalolov, 2024-01-16:
Hello everyone. After proceeding with all the steps described at [https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md) I cannot build the CHIPTool app for android. I have kotlin `Kotlin version 1.7.21-release-272` and did not touch the CHIPTool's files so the gradle version is `gradle-7.1.1` and gradle plugin is `classpath 'com.android.tools.build:gradle:4.2.0'` . But I got this error:

```
> Task :app:compileDebugKotlin FAILED
FAILURE: Build failed with an exception.
28 actionable tasks: 20 executed, 8 up-to-date

* What went wrong:
Execution failed for task ':app:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 32s
```
I don't know where I did wrong. Is there anyone who was able to build CHIPTool app? Thanks

dzulhafiz-at, 2024-05-02:
> @dzulhafiz-at Hello brother, has this problem been solved？

Sorry for the late reply. There was a adjustment on our solution for the app and I had some catching up to do to update the app based on new design. So, I had not revisit this ever since. I couldn't confirm if it works.

Kartikgupta1502, 2024-08-16:
Unable to make field private final java.lang.String java.io.File.path accessible: module java.base does not "opens java.io"
i am facing similar error. How to resolve this any idea?