project-chip/connectedhomeip #28979: [BUG] TC-RVCOPSTATE-2.3 failed when manual step is not supported

pankore, 2023-08-31:
### Reproduction steps

- execute python command (Eg: python3 TC_RVCOPSTATE_2_3.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --storage-path admin_storage.json --PICS ci-pics-values --int-arg PIXIT_ENDPOINT:1 PIXIT_MODEOK:0 PIXIT_MODEFAIL:4 --paa-trust-store-path /paa_root) to test TC-RVCOPSTATE-2.3. 

- It will always fail at Step2, 13: RVCOPSTATE.S.M.ST_PAUSED which required manual steps even if manual steps do not supported.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

2d5fe2ce434f5d0f7714d55e67914e3e8ebf5827

### Platform

ameba

bzbarsky-apple, 2023-08-31:
@duyen-tran please take a look?

plauric, 2023-08-31:
@pankore 

The  RVCOPSTATE.S.M.ST_PAUSED flag indicates whether the DUT supports testing the Paused(0x02) **operational state**. Step 2 of TC-2.3 asks the user to "manually put the DUT into a state wherein it can receive a Pause Command". So this step is agnostic to that flag. 

The user just has to somehow put the device in a state in which it can receive the Pause command: by sending any commands to it using chip-tool, by sending any out of band (i.e. non-Matter) commands, by pressing a physical button on the DUT etc. As long as the user can do that this step will not fail. 

I'll close the issue but please feel to reopen it if you disagree.

closed 2023-08-31 by plauric

pankore, 2023-09-06:
@plauric TC-RVCOPSTATE-2.3 is python scripted, so is there a command used in the docker container to send a command with chip-tool to change the state?
For the meantime, we have no alternative to change the state, and it seems that the script does not read from PICS to skip the manual steps.