project-chip/connectedhomeip #28981: [BUG] TC-IDM-6.2 subscribe request returns INVALID_ACTION

pankore, 2023-08-31:
### Reproduction steps

Using TE2 SHA:5b4f8004662d00bdb111367fec7d3ea978c23372 built chip-all-cluster-app and tested TC-IDM-6.2.
When executing step 2, the response shows INVALID_ACTION.

[tc-idm-6.2.txt](https://github.com/project-chip/connectedhomeip/files/12484112/tc-idm-6.2.txt)

### Bug prevalence

always 

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

other

closed 2023-09-01 by mergify[bot]

pankore, 2023-09-13:
@bzbarsky-apple I found that after applying the commit https://github.com/project-chip/connectedhomeip/commit/5deb59b6caec86fb40726f2574d4bc2d5423106f

In Step 13 and 14, I am still getting INVALID_ACTION after ACL command.
[TC-IDM-6.2-step13-14.txt](https://github.com/project-chip/connectedhomeip/files/12594385/TC-IDM-6.2-step13-14.txt)

bzbarsky-apple, 2023-09-13:
@pankore Thanks!  That's a bug in the test plan....