project-chip/connectedhomeip #28997: [BUG] Removed YAML files were added back in

plauric, 2023-08-31:
### Reproduction steps

This PR removed some YAML files that are superseded by Python test scripts:
https://github.com/project-chip/connectedhomeip/pull/28583

This PR added back the deleted files:
https://github.com/project-chip/connectedhomeip/pull/28829

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

b3c0ac1

### Platform

core

plauric, 2023-08-31:
@manjunath-grl Please help with this.

manjunath-grl, 2023-09-01:
@plauric Removed those tests in PR [28985](https://github.com/project-chip/connectedhomeip/pull/28985) please verify and close the ticket.

plauric, 2023-09-01:
@manjunath-grl Please also remove Test_TC_RVCOPSTATE_2_1.yaml
Test_TC_RVCOPSTATE_2_3.yaml

manjunath-grl, 2023-09-01:
@plauric Above mentioned YAML'S are not there in certification folder. Please verify it once.
https://github.com/manjunath-grl/connectedhomeip/tree/master/src/app/tests/suites/certification

closed 2023-09-01 by plauric