project-chip/connectedhomeip #29007: [BUG] RemoveFabric only make device to remove fabric but device actually don't go commisionable node (_matterc._udp)

qawitherev, 2023-09-01:
### Reproduction steps

device: ESP MoW
ios 16.4 
Device is commissioned using MatterSupport and Matter framework 
1. I decommissioned the device using `MTRBaseClusterOperationalCredentials.removeFabric`
2. Passing in `MTROperationalCredentialsClusterRemoveFabricParams.fabricIndex` which I got from `MTRBaseClusterOperationalCredentials.readAttributeCurrentFabricIndex`. 
3. Was able to remove the fabric, but device stays as as **Operational Device**.

Following is the log from the removeFabric command: 
[image]

c/p: 
I (101528) esp_matter_command: Received command 0x0000000A for endpoint 0x0000's
 cluster 0x0000003E
I (101548) chip[ZCL]: OpCreds: Received a RemoveFabric Command for FabricIndex 0
x2
I (101558) chip[EM]: <<< [E:11312r S:47996 M:96545776 (Ack:116773765)] (S) Msg T
X to 2:238A2C1D7682BB2E [EF04] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (101568) chip[IN]: (S) Sending msg 96545776 on secure session with LSID: 47996
I (101728) chip[FP]: Fabric (0x2) deleted.
I (101738) chip[ZCL]: OpCreds: Fabric index 0x2 was removed
I (101738) chip[DIS]: Updating services using commissioning mode 0
I (101748) chip[DIS]: CHIP minimal mDNS started advertising.
I (101768) chip[DIS]: Advertise operational node 75C5FCE81908DEF0-00000000E12057
72
I (101768) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; inst
ance name: 75C5FCE81908DEF0-00000000E1205772.
I (101788) chip[DIS]: mDNS service published: _matter._tcp
I (101998) chip[ZCL]: OpCreds: RemoveFabric successful
I (101998) chip[SC]: SecureSession[0x3ffc8a38, LSID:47996]: State change 'kActiv
e' --> 'kPendingEviction'
I (102018) chip[EM]: <<< [E:11312r S:47996 M:96545777 (Ack:116773765)] (S) Msg TX to 2:238A2C1D7682BB2E [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (102028) chip[IN]: (S) Sending msg 96545777 on secure session with LSID: 47996
I (102038) app_main: Fabric will be removed
I (102038) app_main: Fabric removed successfully
I (102228) chip[EM]: >>> [E:11312r S:47996 M:116773766 (Ack:96545777)] (S) Msg RX from 2:238A2C1D7682BB2E [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (112468) ROUTE_HOOK: Received RIO
I (112468) ROUTE_HOOK: prefix FDFD:9237:BDAE:: lifetime 1800
I (122398) ROUTE_HOOK: Received RIO
I (122398) ROUTE_HOOK: prefix FDFD:9237:BDAE:: lifetime 1800
I (145848) ROUTE_HOOK: Received RIO
I (145848) ROUTE_HOOK: prefix FDFD:9237:BDAE:: lifetime 1800
I (145848) chip[DL]: IP_EVENT_GOT_IP6
I (145858) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd7a:e4c3:ab96:4f59:266f:28ff:fe87:3288

### Bug prevalence

Every reproduction step 

### GitHub hash of the SDK that was being used

Unsure 

### Platform

darwin, esp32

### Anything else?

Desired output

[image]

What I am trying to achieve, 
1. Unpair the device. 
2. Once unpair, I am able to commission it back to the fabric. 
3. During the [unpaired state] device is still in IP network and is advertising commissionable node service using m]DNS-SD

From my reading, "unpair" device is just simply removing operational credentials from the device so it is removed from the fabric. (Matter Spec 1.0, 6.4.2 - Node Operational Credentials Management),
(Matter Spec 1.0, 4.3.1 - Commissionable Device Discovery)

bzbarsky-apple, 2023-09-01:
Removing the last fabric should generally, per spec:

1. Remove the fabric.
2. Remove any network credentials set up via Matter.

Whether a device then opens a commissioning window at that point depends on the device.  Some devices will do that after a power cycle (the ones that normally enter pairing mode on boot if not commissioned).  Some devices will require other user interaction to open a commissioning window.

That said, in this case the device was a member of two fabrics.  One of the fabrics was removed.  The other one is still there.  So there is no reason whatsoever for the device to open a commissioning window.

closed 2023-09-01 by bzbarsky-apple

qawitherev, 2023-09-04:
> in this case the device was a member of two fabrics

yeah i did notice the removed fabric was 0x2, which means `MTRBaseClusterOperationalCredentials.readAttributeCurrentFabricIndex` returns 0x2 fabric index. In this case it means that my ESP belongs to 2 fabrics and I've only removed the second one so thats why it stays in commissioning mode 0. 

Consider this scenario, 
1. Commissioned a device using MatterSupport and Matter framework and assume device is only on one fabric. 
2. Once commissioned, unpair by using `removeFabric` command, according to your previous reply, it will remove the fabric index inside the device. 
3. Device is still in IP network. To recommission the device, I have to do something to the device [that depends on the device] to open the commissioning window to make it in `commissioning mode 1`, then use Matter framework passing in the 'static' payload (MTRSetupPayload) to commission the device. 

Is this the correct flow?

bzbarsky-apple, 2023-09-05:
> Device is still in IP network.

It shouldn't be, probably.  The spec says to remove any Matter configuration when the last Matter fabric is removed.  But if the device is violating that and staying on the network, then yes, you need to do something device-specific to make it reopen a commissioning window.

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/