project-chip/connectedhomeip #29012: [BUG] TC-SC-4.1 fails to execute FindCommissionable command

rquidute, 2023-09-01:
### Reproduction steps

### Reproduction steps

Step to reproduce:
1. Run TC-SC-4.1 in TH
2. Test fails at step `Step 7a: DNS-SD instance name must be 64-bit randomly selected ID expressed as a sixteen-char hex string with capital letters`  

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

cb83d1ef7376da8e0685e23f5285333c8ce3c4d4

### Platform

python

### Anything else?

This is the request and response for the related test step:

- Request: 
`'json:{ "cluster": "discover", "command": "commissionables", "arguments" : "base64:eyAgfQ==" }'`

- Response: 
`'{  "results": [{ "error": "FAILURE" }],"logs": [{  "module": "TOO",  "category": "Info",  "message": "Q29tbWFuZDogZGlzY292ZXIgY29tbWlzc2lvbmFibGVzIA=="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBjYWNoZSBleGhhdXN0ZWQ="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBhbGwgZm9yIG5vdw=="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBjYWNoZSBleGhhdXN0ZWQ="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBhbGwgZm9yIG5vdw=="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBjYWNoZSBleGhhdXN0ZWQ="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBhbGwgZm9yIG5vdw=="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBjYWNoZSBleGhhdXN0ZWQ="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBhbGwgZm9yIG5vdw=="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBjYWNoZSBleGhhdXN0ZWQ="},{  "module": "DL",  "category": "Info",  "message": "QXZhaGkgYnJvd3NlOiBhbGwgZm9yIG5vdw=="}]}'`

- This is the step in YAML script:
```
    - label:
          "Step 7a: DNS-SD instance name must be 64-bit randomly selected ID
          expressed as a sixteen-char hex string with capital letters"
      cluster: "DiscoveryCommands"
      command: "FindCommissionable"
      response:
          values:
              - name: "instanceName"
                constraints:
                    minLength: 16
                    maxLength: 16
                    isUpperCase: true
                    isHexString: true
```

bzbarsky-apple, 2023-09-01:
Those logs are "Avahi browse: cache exhausted" and "Avahi browse: all for now" repeated a bunch of times.

It sounds like the actual discovery is failing for some reason in your setup...  Could you please attach (not paste) a complete log here, from both the TH and the device involved @rquidute ?

rquidute, 2023-09-01:
Ho @bzbarsky-apple, I'm sharing the requested logs bellow:

[DUT_Log.zip](https://github.com/project-chip/connectedhomeip/files/12500571/DUT_Log.zip)
[UI_Test_Run_2023_09_01_14_47_30.log.zip](https://github.com/project-chip/connectedhomeip/files/12500572/UI_Test_Run_2023_09_01_14_47_30.log.zip)

bzbarsky-apple, 2023-09-01:
@rquidute This part from the TH log:
```
INFO       | 2023-09-01 17:51:44.327595 | Executing Test Step: Step 6: DUT is Commissioned and instructed to enter in commissioning mode using Open Commissioning Window command
WARNING    | 2023-09-01 17:52:22.101676 | Test Failure: User stated manual step FAILED.
WARNING    | 2023-09-01 17:52:22.102702 | Test Failure: Manual Test Step Failure.
INFO       | 2023-09-01 17:52:22.118708 | Test Step Completed [FAILED]: Step 6: DUT is Commissioned and instructed to enter in commissioning mode using Open Commissioning Window command
```

It looks like for non-CI the test expects a manual OpenCommissioningWindow command to be sent. Was that done in this case?

rquidute, 2023-09-01:
@bzbarsky-apple Let me share my execution in this step:
[DUT-step6.zip](https://github.com/project-chip/connectedhomeip/files/12500794/DUT-step6.zip)

bzbarsky-apple, 2023-09-01:
@rquidute Well, that log shows the command failed, right?  So the commissioning window is not open, and there is nothing to discover, etc.

The command you are running is:
```
chip-tool pairing open-commissioning-window 1 0 200 2000 3840
```
which is (1) doing OpenBasicCommissioningWindow, not OpenCommissioningWindow and (2) using `1` as the node id for the server.  But from the TH log:

```
INFO       | 2023-09-01 17:47:31.869178 | New Node Id generated: 0x830fa33d212bd82f
...
CHIPTOOL   | 2023-09-01 17:47:33.872551 | CHIP:TOO Command: pairing onnetwork-long 0x830fa33d212bd82f 20202021 3840 --trace_file "/logs/trace_log_2023-09-01_17.47.32_0x830fa33d212bd82f_PAIRING_onnetwork-long.log" --trace_decode 1 
```

and from the device log:

```
[1693590461.578185][170942:170942] CHIP:DIS: Advertise operational node FC0B00724FC4B32C-830FA33D212BD82F
```

(and similar for all the other "Advertise operational node" bits).

So the node is commissioned with node ID `0x830fa33d212bd82f`, not `1`.

closed 2023-09-01 by bzbarsky-apple