project-chip/connectedhomeip #29031: [BUG]

LongND0105, 2023-09-02:
### Reproduction steps

I have a lighting app for esp32 which included extended color cluster, I'm using chip-tool on ubuntu 22.04 to pairing the esp32.
I want to subscribe to onoff cluster and levelcontrol cluster but I only subscribe to one cluster at a endpoint.
1. Subscribe to onoff attribute on onoff cluster: It return attribute value when I send onoff on/off/toggle command or press button on esp32.
2. Subscribe to current-level attribute on levelcontrol cluster: It return attribue value
3. Now, I'm resending onoff command or press button on esp32 but chip-tool not receive report from esp32.

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

I don't know

### Platform

other

### Platform Version(s)

1.0

bzbarsky-apple, 2023-09-02:
@LongND0105 When you did the second subscribe, did you pass `--keepSubscriptions true`?  The default behavior of a subscribe is to cancel other existing subscriptions.

LongND0105, 2023-09-04:
I did not pass `--keepSubscriptions true` to the command subscribe

bzbarsky-apple, 2023-09-05:
Then the behavior is all as expected.

closed 2023-09-05 by bzbarsky-apple