project-chip/connectedhomeip #29033: [BUG] Address review issues on PR 28836

bzbarsky-apple, 2023-09-02:
### Reproduction steps

See https://github.com/project-chip/connectedhomeip/pull/28836#pullrequestreview-1607869620 @lpbeliveau-silabs 

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

1ed389725756a7a5e6ce4c2e79e6458c7d89d2f6

### Platform

core

bzbarsky-apple, 2023-09-07:
Note: https://github.com/project-chip/connectedhomeip/issues/29033 did not fully address this, both because it ignored https://github.com/project-chip/connectedhomeip/pull/28836#discussion_r1313709012 and because of https://github.com/project-chip/connectedhomeip/pull/29076#discussion_r1318849226 @andy31415 @jmartinez-silabs

lpbeliveau-silabs, 2023-10-02:
Linking the spec issue opened here: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/7536
as well as the null level specific issue opened here: https://github.com/project-chip/connectedhomeip/issues/29185

closed 2023-11-23 by mergify[bot]