project-chip/connectedhomeip #29039: [BUG] crash ：ChipDeviceControllerException constructor  not found

netscell, 2023-09-03:
### Reproduction steps

call pair device，throw ChipDeviceControllerException and android app crashed：

the android crash log：
java.lang.NoSuchMethodError: no non-static method "Lchip/devicecontroller/ChipDeviceControllerException;.<init>(ILjava/lang/String;)V"
	at chip.devicecontroller.ChipDeviceController.pairDevice(Native Method)
	at chip.devicecontroller.ChipDeviceController.pairDevice(ChipDeviceController.java:160)
	at chip.devicecontroller.ChipDeviceController.pairDevice(ChipDeviceController.java:129)

the ThrowError in JniReferences.cpp  get method is ：
jmethodID constructor = env->GetMethodID(exceptionCls, "<init>", "(ILjava/lang/String;)V");
but in ChipDeviceControllerException.java constructor param errorCode type is long，not int；

### Bug prevalence

1-2times a day

### GitHub hash of the SDK that was being used

sdk version 1.0.0.2

### Platform

android

yunhanw-google, 2023-09-04:
@netscell 
this issue should be fixed by https://github.com/project-chip/connectedhomeip/pull/26802
please use the latest code, thanks

joonhaengHeo, 2023-09-04:
@yunhanw-google 
However, some code is creating using JniReference::GetInstance().ThrowError()

https://github.com/project-chip/connectedhomeip/blob/54038c0db7349cb54461dbb59418674d747123cf/src/controller/java/CHIPDeviceController-JNI.cpp#L129

https://github.com/project-chip/connectedhomeip/blob/54038c0db7349cb54461dbb59418674d747123cf/src/controller/java/CHIPDeviceController-JNI.cpp#L142

yunhanw-google, 2023-09-04:
yep, agree, ThrowError is still problematic to handle the error with uint32_t, let's fix that, thanks

PR #29066 (2023-09-05): [Android] Fix java exception type
Fix #29039 

Java exception errorCode type is changed to long type.
For this reason, errorCode type is defined uint32_t.