project-chip/connectedhomeip #29040: [BUG] crash NsdServiceFinderAndResolver.onServiceFound throw java.lang.NullPointerException

netscell, 2023-09-03:
### Reproduction steps

control device occurs crash sometime ,the android log :

java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.concurrent.ScheduledFuture.cancel(boolean)' on a null object reference
	at chip.platform.NsdServiceFinderAndResolver.onServiceFound(NsdServiceFinderAndResolver.java:102)
	at android.net.nsd.NsdManager$ServiceHandler.lambda$handleMessage$2(NsdManager.java:696)
	at android.net.nsd.NsdManager$ServiceHandler$$ExternalSyntheticLambda4.run(Unknown Source:4)
	at android.net.nsd.NsdManager$$ExternalSyntheticLambda0.execute(Unknown Source:0)
	at android.net.nsd.NsdManager$ServiceHandler.handleMessage(NsdManager.java:696)
	at android.os.Handler.dispatchMessage(Handler.java:109)
	at android.os.Looper.loopOnce(Looper.java:206)
	at android.os.Looper.loop(Looper.java:296)
	at android.os.HandlerThread.run(HandlerThread.java:67)
-loghandler

### Bug prevalence

1-2times a day

### GitHub hash of the SDK that was being used

sdk version 1.0.0.2

### Platform

android

joonhaengHeo, 2023-09-04:
@netscell 
Please upload the entire log related to the issue.

netscell, 2023-11-22:
encounter this crash when stress testing with a matter light bulb：
java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.concurrent.ScheduledFuture.cancel(boolean)' on a null object reference
	at chip.platform.NsdServiceFinderAndResolver.onServiceFound(NsdServiceFinderAndResolver.java:102)
	at android.net.nsd.NsdManager$ServiceHandler.lambda$handleMessage$2(NsdManager.java:863)
	at android.net.nsd.NsdManager$ServiceHandler$$ExternalSyntheticLambda10.run(Unknown Source:4)
	at android.net.nsd.NsdManager$$ExternalSyntheticLambda0.execute(Unknown Source:0)
	at android.net.nsd.NsdManager$ServiceHandler.handleMessage(NsdManager.java:863)
	at android.os.Handler.dispatchMessage(Handler.java:109)
	at android.os.Looper.loopOnce(Looper.java:206)
	at android.os.Looper.loop(Looper.java:296)
	at android.os.HandlerThread.run(HandlerThread.java:67)

kuyu132, 2023-12-26:
I faced the same issue, the sdk version I used is 1.2.0
--------- beginning of crash
12-25 23:38:42.609 30402 30927 E AndroidRuntime: FATAL EXCEPTION: NsdManager
12-25 23:38:42.609 30402 30927 E AndroidRuntime: Process: com.test, PID: 30402
12-25 23:38:42.609 30402 30927 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.concurrent.ScheduledFuture.cancel(boolean)' on a null object reference
12-25 23:38:42.609 30402 30927 E AndroidRuntime: 	at chip.platform.NsdServiceFinderAndResolver.onServiceFound(NsdServiceFinderAndResolver.java:102)
12-25 23:38:42.609 30402 30927 E AndroidRuntime: 	at android.net.nsd.NsdManager$ServiceHandler.handleMessage(NsdManager.java:416)
12-25 23:38:42.609 30402 30927 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
12-25 23:38:42.609 30402 30927 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:201)
12-25 23:38:42.609 30402 30927 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:288)
12-25 23:38:42.609 30402 30927 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:67)
12-26 00:35:03.221 20690 21257 E AndroidRuntime: FATAL EXCEPTION: NsdManager
12-26 00:35:03.221 20690 21257 E AndroidRuntime: Process: com.test, PID: 20690
12-26 00:35:03.221 20690 21257 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.concurrent.ScheduledFuture.cancel(boolean)' on a null object reference
12-26 00:35:03.221 20690 21257 E AndroidRuntime: 	at chip.platform.NsdServiceFinderAndResolver.onServiceFound(NsdServiceFinderAndResolver.java:102)
12-26 00:35:03.221 20690 21257 E AndroidRuntime: 	at android.net.nsd.NsdManager$ServiceHandler.handleMessage(NsdManager.java:416)
12-26 00:35:03.221 20690 21257 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
12-26 00:35:03.221 20690 21257 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:201)
12-26 00:35:03.221 20690 21257 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:288)
12-26 00:35:03.221 20690 21257 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:67)
12-26 00:54:23.467 27831 28340 E AndroidRuntime: FATAL EXCEPTION: NsdManager

Trishia, 2024-01-05:
I have faced same issue with the sdk 1.2.0, advise me how to this solve. 

--------- beginning of crash
17:16:16.858  5988-6001  DMG                      D  ICR moving to [AwaitingDe]
17:16:16.859  5988-6046  AndroidRuntime           E  FATAL EXCEPTION: NsdManager
                                                     Process: com.xxx.xxx, PID: 5988
                                                     java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.concurrent.ScheduledFuture.cancel(boolean)' on a null object reference
                                                     	at chip.platform.NsdServiceFinderAndResolver.onServiceFound(NsdServiceFinderAndResolver.java:102)
                                                     	at android.net.nsd.NsdManager$ServiceHandler.handleMessage(NsdManager.java:387)
                                                     	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                     	at android.os.Looper.loop(Looper.java:193)
                                                     	at android.os.HandlerThread.run(HandlerThread.java:65)
17:16:16.870  5988-6046  Process                  I  Sending signal. PID: 5988 SIG: 9

closed 2024-02-23 by mergify[bot]