project-chip/connectedhomeip #29054: [CERT-TEST-FAILURE]

Saravana-kr22, 2023-09-04:
### Feature Area

Other

### Test Case

TC-IDM-10.1

### Reproduction steps

Command to run the Test-case script: `python3 TC_IDM_10_1.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --storage-path admin_storage.json --PICS ci-pics-values --paa-trust-store-path /paa_roots`

### Bug prevalence

Whenever i do

### GitHub hash of the SDK that was being used

54038c0db7349cb54461dbb59418674d747123cf

### Platform

python, raspi

### Anything else?

**Description:** 
- This Test case has 2 python file for [TC_IDM_10_1.py](https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_IDM_10_1.py) and the [TC_DeviceBasicComposition.py](https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_DeviceBasicComposition.py)
- The [TC_IDM_10_1.py](https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_IDM_10_1.py) is missed to verify the step-3: `For every cluster on every endpoint verify that the following global attributes do not contain duplicate values` and Step-4:`Validate all reported attributes match AttributeList` of the test-plan.
- The [TC_DeviceBasicComposition.py](https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_DeviceBasicComposition.py) is missed to verify the newly added test-step in the latest test-plan.

**Reference:**
Test-plan: https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-10-1-cluster-requirements-global-attributes-dut_server

**Logs:**

TC_IDM_10_1.py - [tc-idm-10-1.txt](https://github.com/project-chip/connectedhomeip/files/12513636/tc-idm-10-1.txt)

cecille, 2023-09-08:
Is this a test failure? Or just a concern about missing steps? The test in device basic composition verifies no duplicates. @ReneJosefsen - does the named test verify 6, 7, 8 in the test plan? If so, together these tests cover the entire test plan.

ReneJosefsen, 2023-09-08:
The current setup is that both test needs to be executed and combined the aim is to get full coverage. Actually I think they overlap in some cases.

I agree this is a bit confusing as is, and ideally these scripts should be combined, but we didn't manage to get that done in time for the SDK cut. We are already working on getting them merged.

closed 2024-01-24 by Saravana-kr22