project-chip/connectedhomeip #29062: Why does the last parameter of esp_ble_gatts_send_indicate use false?

nykh2010, 2023-09-05:
https://github.com/project-chip/connectedhomeip/blob/eb019b0f150cf31becb6252709c389c4ae297190/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp#L855

In a noisy environment, it is easy to lost BTP packet when transmit PAI or DAC cert. It will cause the commission process to fail. 
when I set the last parameter to true, the commission process always successfully passed.  so the last parameter needs to set true?

the IDF version is 4.4.5, platform is esp32c3, connectedhomeip tag 1.1.0.1

bzbarsky-apple, 2023-09-05:
Looks like exactly the thing that https://github.com/project-chip/connectedhomeip/pull/28923 fixes.

bzbarsky-apple, 2023-09-07:
The PR is not merged....

closed 2023-09-11 by mergify[bot]