project-chip/connectedhomeip #29069: [BUG]

nicelyjust, 2023-09-05:
### Reproduction steps

SOE:MEIZU  Android 7
when subscribe the attributes，
`     val endpointId = ChipPathId.forId(endpoint)
      val clusterId = ChipPathId.forWildcard()
      val attributeId = ChipPathId.forWildcard()
      val minInterval = 1 // seconds
      val maxInterval = 4 // seconds
      val attributePath = ChipAttributePath.newInstance(endpointId, clusterId, attributeId)
      val eventPath = ChipEventPath.newInstance(endpointId, clusterId, attributeId)
      TLog.d(TAG,"attributePath: [${attributePath}]")
       chipClient.chipDeviceController.subscribeToPath(...)`
then app will crash!
`A  art/runtime/indirect_reference_table.cc:132] JNI ERROR (app bug): local reference table overflow (max=512)
A  art/runtime/indirect_reference_table.cc:132] local reference table dump:
A  art/runtime/indirect_reference_table.cc:132]   Last 10 entries (of 512):
A  art/runtime/indirect_reference_table.cc:132]       511: 0x1386cad8 java.lang.String "{"value":[]}"
A  art/runtime/indirect_reference_table.cc:132]       510: 0x132a8570 byte[] (2 elements)
A  art/runtime/indirect_reference_table.cc:132]       509: 0x1375cc58 java.util.ArrayList
A  art/runtime/indirect_reference_table.cc:132]       508: 0x1375ccb8 chip.devicecontroller.model.ChipAttributePath
A  art/runtime/indirect_reference_table.cc:132]       507: 0x13904f00 java.lang.Class<chip.devicecontroller.model.NodeState>
A  art/runtime/indirect_reference_table.cc:132]       506: 0x1375cd90 chip.devicecontroller.model.AttributeState
A  art/runtime/indirect_reference_table.cc:132]       505: 0x13868de0 java.lang.Class<chip.devicecontroller.model.AttributeState>
A  art/runtime/indirect_reference_table.cc:132]       504: 0x13cc2ab8 java.lang.String "chip/devicecontr... (42 chars)
A  art/runtime/indirect_reference_table.cc:132]       503: 0x136506d0 java.lang.String "chip/devicecontr... (45 chars)
A  art/runtime/indirect_reference_table.cc:132]       502: 0x13904600 java.lang.Class<chip.devicecontroller.model.ChipAttributePath>
A  art/runtime/indirect_reference_table.cc:132]   Summary:
A  art/runtime/indirect_reference_table.cc:132]       223 of java.lang.Class (11 unique instances)
A  art/runtime/indirect_reference_table.cc:132]        90 of java.lang.String (89 unique instances)
A  art/runtime/indirect_reference_table.cc:132]        77 of java.lang.Long (77 unique instances)
A  art/runtime/indirect_reference_table.cc:132]        40 of chip.devicecontroller.model.ChipAttributePath (40 unique instances)
A  art/runtime/indirect_reference_table.cc:132]        39 of chip.devicecontroller.model.AttributeState (39 unique instances)
A  art/runtime/indirect_reference_table.cc:132]        16 of java.util.ArrayList (16 unique instances)
A  art/runtime/indirect_reference_table.cc:132]        15 of java.lang.Integer (15 unique instances)
A  art/runtime/indirect_reference_table.cc:132]         6 of java.lang.Boolean (6 unique instances)
A  art/runtime/indirect_reference_table.cc:132]         2 of chip.devicecontroller.model.NodeState (2 unique instances)
A  art/runtime/indirect_reference_table.cc:132]         2 of java.util.HashMap (2 unique instances)
A  art/runtime/indirect_reference_table.cc:132]         1 of chip.devicecontroller.ChipStructs$DescriptorClusterDeviceTypeStruct
A  art/runtime/indirect_reference_table.cc:132]         1 of byte[] (2 elements)
A  art/runtime/indirect_reference_table.cc:132] 
`

### Bug prevalence

always

### GitHub hash of the SDK that was being used

8db0d655a074c8b3d5539c7847fc54f8448cb5a3

### Platform

android

### Platform Version(s)

Android 7 cpu:qualcomm 625

### Anything else?

[native crash.txt](https://github.com/project-chip/connectedhomeip/files/12521894/native.crash.txt)

nicelyjust, 2023-09-05:
https://stackoverflow.com/questions/59680110/crash-error-jni-error-local-reference-table-overflow-max-512 , maybe This is why Google Matter supports a minimum of version 8.0

yunhanw-google, 2023-09-08:
@nicelyjust could you trying the latest master code, and see whether this issue can be reproduced?

yunhanw-google, 2023-09-12:
@nicelyjust  are you in slack? could you find me in yunhanw-google in slack?
Could you also try this fix https://github.com/project-chip/connectedhomeip/pull/29175, the local reference issue spreads in multiple place, this PR should help ChipAttributePath entry in local reference tag per your log.

nicelyjust, 2023-09-13:
@yunhanw-google Thank for your response.I'll try it later, it needs to be compiled

yunhanw-google, 2023-09-13:
@nicelyjust please hold on a little for testing. We just root caused the leak points, and would create the pull request in the morning, we would also have another 2 or 3 pull request to fix other located leaking local reference points.

nicelyjust, 2023-09-13:
@yunhanw-google yes,I'm.  Already sent you a message on Slack

closed 2023-11-06 by yunhanw-google