project-chip/connectedhomeip #29079: [BUG] How to have Contact Sensor as Bridged device in matter bridge

mttrRDRSDX, 2023-09-06:
### Reproduction steps

currently in example code we have light , temp sensor composed deivce etc as a bridged device

how can i have contact sensor like that 
as i can not see any class for contact sensor in src/app/clusters/

### Bug prevalence

Whenever

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/master/src/app/clusters

### Platform

core

bzbarsky-apple, 2023-09-06:
For a contact sensor, the only things to implement are the Boolean State cluster and Identify.   Identify does have a class for it.  Boolean State just requires implementing the StateValue attribute, via the external attribute read/write callbacks.

closed 2023-09-06 by bzbarsky-apple

mttrRDRSDX, 2023-09-07:
From creating cluster , 
for non-matter device 
DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN MACROS should be fine right.
or zaptool should come into the picture ?

bzbarsky-apple, 2023-09-07:
DECLARE_DYNAMIC_CLUSTER_* macros should generally be what you want here.

mttrRDRSDX, 2023-09-08:
hey 
thanks for your support 

i wrote the DYNAMIC EP as follows 

/*Contact Sensor*/
DeviceCT ContactSensor("Contact Sensor", "LivingArea");

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(contactSensorAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(ContactSensor.Id, BOOLEAN, 1, ZAP_ATTRIBUTE_MASK(WRITABLE) | ZAP_ATTRIBUTE_MASK(NULLABLE)),

//DECLARE_DYNAMIC_ATTRIBUTE(),
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedContactSensorClusters)
DECLARE_DYNAMIC_CLUSTER(ContactSensor.AttrId, contactSensorAttrs, nullptr, nullptr),
DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs, nullptr, nullptr),
// DECLARE_DYNAMIC_CLUSTER();
// DECLARE_DYNAMIC_CLUSTER(ZCL_BRIDGED_DEVICE_BASIC_CLUSTER_ID, bridgedDeviceBasicAttrs, nullptr, nullptr))
DECLARE_DYNAMIC_CLUSTER_LIST_END;

DECLARE_DYNAMIC_ENDPOINT(bridgedContactSensorEndpoint, bridgedContactSensorClusters);
DataVersion gContactSensorDataVersions[ArraySize(bridgedContactSensorClusters)];

it creates an Endpoint successfully on call to AddEndpoint API 
but when i try to read from the chiptool 
via this command 
**bridgeddevicebasicinformation read node-label 1 3**

rather then showing the contact sensor name 
it shows the CLUSTER NOT_SUPPORTED 
**[1694157592.128491][347455:347457] CHIP:DMG: 				StatusIB =
[1694157592.128522][347455:347457] CHIP:DMG: 				{
[1694157592.128550][347455:347457] CHIP:DMG: 					status = 0xc3 (UNSUPPORTED_CLUSTER),
[1694157592.128579][347455:347457] CHIP:DMG: 				},
[1694157592.128608][347455:347457] CHIP:DMG:** 				

it seems i am missing something can you please help in finding 
FYI 
DeviceCT is the new class in Device.h / .cpp which i wrote for contact sensor

bzbarsky-apple, 2023-09-08:
Well, you don't have a Bridged Device Basic Information cluster there, right?  You have Descriptor, and you have ContactSensor (though the "AttrId" thing is not making sense to me), but where would Bridged Device Basic Information come from given those decls?

mttrRDRSDX, 2023-09-09:
### Well, you don't have a Bridged Device Basic Information cluster there, right? You have Descriptor, and you have ContactSensor

Yes i tried Adding the BridgedDeviceBasicInformation cluster worked fine 

now expected results are coming as follows : 

**[1694240336.865654][429728:429730] CHIP:DMG:                                    Endpoint = 0x4,
[1694240336.865680][429728:429730] CHIP:DMG:                                    Cluster = 0x39,
[1694240336.865706][429728:429730] CHIP:DMG:                                    Attribute = 0x0000_0005,
[1694240336.865731][429728:429730] CHIP:DMG:                            }
[1694240336.865759][429728:429730] CHIP:DMG:
[1694240336.865787][429728:429730] CHIP:DMG:                            Data = "Contact Sensor" (14 chars),
[1694240336.865811][429728:429730] CHIP:DMG:                    },
[1694240336.865840][429728:429730] CHIP:DMG:
[1694240336.865863][429728:429730] CHIP:DMG:            },
[1694240336.865894][429728:429730] CHIP:DMG:
[1694240336.865915][429728:429730] CHIP:DMG:    ],
[1694240336.865946][429728:429730] CHIP:DMG:**

### (though the "AttrId" thing is not making sense to me),
 

DECLARE_DYNAMIC_ATTRIBUTE(ContactSensor.Id, BOOLEAN, 1, ZAP_ATTRIBUTE_MASK(WRITABLE) | ZAP_ATTRIBUTE_MASK(NULLABLE)),

ContactSensor.Id is only fine to reflect the StateValue of the Contact Sensor right

mttrRDRSDX, 2023-09-14:
am waiting for response any update ?would be helpful 
as i am stuck in adding the commands like 
C T open <1> <1> like this 
please check my understanding in last post and if you can contrinute in any way .

bzbarsky-apple, 2023-09-14:
At this point I am very unclear on what the question, if any, is here.  Can you clearly state that part?

mttrRDRSDX, 2023-09-15:
let me share the scenerio i am in 
i have added a CT device in matter bridge 
and matter bridge is commissioned to google nest 

now just like we can do on / off operation and we can  see the status of the light app
i wanted it same for CT 

in order to have the attribute status and all in google nest for CT device i digged into the code 
where i could found that in order to create any command -> onoff toggle 1 1
we have some classes defined in this file  src/app/clusters/ 
as well as some python code also i was able to found related to the attribute / object 

now query is simple
if i want to do some operation or command from chip-tool / google nest to open close the CT device for the simulation purpose only 
i need something like this -> CT open 1 1 
correct?

so in order to do these changes i digged the code but its not clear how these commands for any example were introduced 

something which is used in this array  in matter bridge 

constexpr CommandId onOffIncomingCommands[] = {
    app::Clusters::OnOff::Commands::Off::Id,
    app::Clusters::OnOff::Commands::On::Id,
    app::Clusters::OnOff::Commands::Toggle::Id,
    app::Clusters::OnOff::Commands::OffWithEffect::Id,
    app::Clusters::OnOff::Commands::OnWithRecallGlobalScene::Id,
    app::Clusters::OnOff::Commands::OnWithTimedOff::Id,
    kInvalidCommandId,
};

if i wwant to do it for Ct device 
requirement :
constexpr CommandId openCloseIncomingCommands[] = {
    -----
};

how can i have the open close commands for my bridged ct device ?
changing only src/app/cluster doesnt works it seems we need some python chages as well?

bzbarsky-apple, 2023-09-15:
So you want to add a new on-the-wire command that is not part of the spec?

You'd need to change the XML for the cluster to define the command, etc.  And give it a vendor-scoped ID, presumably.

mttrRDRSDX, 2023-09-16:
so following this below example commit would be fine right ?

https://github.com/project-chip/connectedhomeip/pull/26572

an example of refrigerator app

bzbarsky-apple, 2023-09-17:
@mttrRDRSDX Depending on what you are trying to do, yes.

rajeshkrishresi, 2023-09-27:
Hey i have one last query 

i have a current setup ready as 

Non-matter network Device  <-> matter bridge <-> google nest

so as a non matter network device i have -> Contact sensor
and i want when ever i open close Contact sensor , matter light should be toggled 

so for that do we need some kind of binding command ?
or it can be done at code level as well ? if so any pointer you can give ?

mttrRDRSDX, 2023-09-27:
Hi, 

i have one last query 
i have a setup ready as 
non-matter-network <-> matter bridge <-> Google nest

to be more precise
non-matter-network device as -> Contact sensor
Google nest had light app matter

how can i bind contact sensor to matte light ?
do we need to do something explicitly from chip-tool?
or we can do it from code as well dynamically?

to make sure whenever contact sensor open / close , matter light toggles .

bzbarsky-apple, 2023-09-28:
> to make sure whenever contact sensor open / close , matter light toggles .

That needs something to observe the state of the sensor and send the relevant commands to the light, right @mttrRDRSDX ?

bzbarsky-apple, 2023-09-28:
> whether matter bridge has some way to look into the matter network and get list of all endpoints avail?

For a given _node_, if you have the right access, you can get the list from PartsList on Descriptor on endpoint 0 of that node.

You can do a DNS-SD browse for `_matter._tcp` to find all the operational advertisements, which should let you find out what nodes are around.

mttrRDRSDX, 2023-09-28:
You can do a DNS-SD browse for _matter._tcp to find all the operational advertisements, which should let you find out what nodes are around.

**DNS-SD browse give me _matter._tcp this part is fine 
to find what all nodes are around** 

For a given node, if you have the right access, you can get the list from PartsList on Descriptor on endpoint 0 of that node.

**Again in order to get the partsList we need to access the node first right ?
but still am bit confused on part accessing the node 
how this can be automated in the matter bridge ?
there is some DNS discovery in matter bridge also will it be helpful to find ?
like
 we have matter smoke sensor , matter light , matter switch around .
matter bridge will then store those node details and then able to toggle the light .**

bzbarsky-apple, 2023-09-28:
You can't really discover what a node implements via DNS-SD discovery.  You can discover nodes, connect to them, and then read their capabilities....

mttrRDRSDX, 2023-10-03:
I tried using chip tool with both the following commands :

**> discover commissionables**
it shows nothing 

then 

**> discover commissioners**

it is able to find the bridge app only 

but as discussed already 
i want to find the light app which is commissioned to google nest

is it a correct way ?

With avahi i am able to see below nodes
+ wlp2s0 IPv6 dca907a71dfaa5ea-000000009d125602             _matter._tcp         local
+ wlp2s0 IPv6 DCA907A71DFAA5EA-00000000931E0C96             _matter._tcp         local
+ wlp2s0 IPv6 DCA907A71DFAA5EA-00000000030CBC08             _matter._tcp         local
+ wlp2s0 IPv4 dca907a71dfaa5ea-000000009d125602             _matter._tcp         local
+ wlp2s0 IPv4 DCA907A71DFAA5EA-00000000931E0C96             _matter._tcp         local

now the problem is how can i look into them access them ?
via chiptool ?

and if so then how come bridge will be able to send commands to these nodes?
do we need some kind of binding?

bzbarsky-apple, 2023-10-04:
>  i want to find the light app which is commissioned to google nest

Then you want to find an operational node, not a commissionable one?   In that case "discover commissionables" won't do what you want....

> With avahi i am able to see below nodes

Yes, those are operational advertisements, looks like for three separate nodes all on the same fabric.  One of which is violating the spec for some reason (whichever one is advertising as `dca907a71dfaa5ea-000000009d125602`) by not having the right casing on its advertisements.

> now the problem is how can i look into them access them ?

Well, those nodes are all on the same fabric (probably; the compressed fabric id could collide, but it's not likely).  You need something that is on that fabric to access them.  Were these nodes commissioned by chip-tool?  If not, it doesn't have the security material to talk to them, right?

> and if so then how come bridge will be able to send commands to these nodes?

Is the bridge on the relevant fabric?  If so, it has the relevant security material.

> i couldn't find a way to connect matter bridge to matter device 

This needs to be done by something that is an administrator for both devices.

mttrRDRSDX, 2023-10-04:
**Were these nodes commissioned by chip-tool? If not, it doesn't have the security material to talk to them, right?**

No none of the nodes were commissioned to chip-tool apart from matter bridge app 

**Is the bridge on the relevant fabric? If so, it has the relevant security material.**

yes bridge is on fabric
sorry but i didnt understood the security material part

**This needs to be done by something that is an administrator for both devices.**

okay , can we consider the google nest as the administrator?
since it is the controller and has both bridge-app (bridged devices)  and the matter lightapp.

bzbarsky-apple, 2023-10-04:
>  sorry but i didnt understood the security material part

To communicate with a Matter node you either have to send it a group message (which requires both nodes to have the same group encryption/decryption key) or send it a unicast encrypted message (which requires both nodes to have certificates chaining up to the same root key, and to have the same Identity Protection Key).

> any way of doing this

The avahi browse above is the right way: you do a dns-sd browse for `_matter._tcp`.  Or the equivalent via any other dns-sd client...

chip-tool does not have a built-in command to discover operational things; it hasn't come up, so no one implemented it.

mttrRDRSDX, 2023-10-04:
**The avahi browse above is the right way:** 

okay thanks , i will do this way 
but more i was interested was in having some run-time operational discovery , bridge will perform at runtime and it will take decision based on logic whether to connect to that node or not .

if there is no support for operation discovery then i will try for some workaround or use suggested method 

**To communicate with a Matter node you either have to send it a group message (which requires both nodes to have the same group encryption/decryption key) or send it a unicast encrypted message (which requires both nodes to have certificates chaining up to the same root key, and to have the same Identity Protection Key).**

do we have any example for doing the steps ?
as it requires the crypto thing , so margin of error should be zero.

bzbarsky-apple, 2023-10-05:
> but more i was interested was in having some run-time operational discovery

https://github.com/project-chip/connectedhomeip/pull/26718 and https://github.com/project-chip/connectedhomeip/pull/28301 for some code in that general direction, but at the moment there is nothing cross-platform in the SDK for this.

> do we have any example for doing the steps ?

Doing what steps?